SpikeInterface/spikeinterface #2449: export_to_phy with amplitudes and principal components

Ashkees, 2024-01-30:
Hello,

I'd like to export to Phy and be able to visualize the amplitudes and principal components.  I'm able to run the following code without errors, but when I load into Phy, the amplitudes and principal components aren't there.
```
we = si.extract_waveforms(recording_loaded, sorting_SC2_loaded, folder=base_folder / "waveforms_sparse",
                              sparse=True, overwrite=True, **job_kwargs)

from spikeinterface.postprocessing import compute_spike_amplitudes, compute_principal_components
from spikeinterface.exporters import export_to_phy

_ = compute_spike_amplitudes(waveform_extractor=we)
_ = compute_principal_components(waveform_extractor=we, n_components=3, mode='by_channel_global')

si.export_to_phy(we, output_folder=base_folder / 'phy_SC2', remove_if_exists=True,
                           compute_amplitudes=False, compute_pc_features=False, copy_binary=False,
                           **job_kwargs)
```
Alternatively, if I calculate the amplitudes and PCs while exporting (compute_amplitudes=True, compute_pc_features=True), this works fine.  But I gather from the documentation that this isn't a great idea.

Thanks in advance for the advice!
Ashley

alejoe91, 2024-01-30:
Hi, you just need to set the `compute_amplitudes` and `compute_principal_components` to `True` in the `export_to_phy` function

Ashkees, 2024-01-30:
Ok, thank you! This works fine.  The documentation just made it sound like it was better to calculate it before, which was why I was trying to make it work that way.

alejoe91, 2024-01-30:
So for the `compute_amplitudes`, they are already calculates, so they won't be computed again.

The PCA deserves a different explanation: by default, the postprocessing function fits the PCA model and projects only the pre-extracted waveforms (from the `WaveformExtractor`). Phy actually needs all projections, so the projections for all spikes are run on the fly!

I agree this is probably not super clear from the docs!

Ashkees, 2024-01-30:
Ah, ok, I understand now.  Thanks!

closed 2024-01-30 by alejoe91