SpikeInterface/spikeinterface #2452: Error in run_sorter : file spikeinterface_log.json empty

Mymi-INS, 2024-01-31:
Hi everyone,

I have an issue when running my SI pipeline. 
The error is JSONDecodeERROR : Expecting value : line 1 column 1 (char0). I checked and it's because the file spikeinterface_log.json is empty. I don't understand why it's empty cause every other file that SI create like spikeinterface_recording.json and _params.json are not empty. 
Can you help me with that please?

Thanks :) 

Myriam

alejoe91, 2024-01-31:
Hi @Mymi-INS 

Can you print the output of your script? Where is the error triggered?

alejoe91, 2024-01-31:
What is the output before that? Can you also share the version of SpikeInterface that you are using?

Mymi-INS, 2024-01-31:
What is written before is that : 

Spike sorting launched at 2024-01-30 12:32:37.567109
mypath/SpikeInterface_output
Starting container
Installing spikeinterface==0.99.1 in spikeinterface/kilosort2_5-compiled-base
Running kilosort2_5 sorter inside spikeinterface/kilosort2_5-compiled-base
Stopping container
Traceback (most recent call last):

and I guess my SI version is 0.99.1

alejoe91, 2024-01-31:
did you notice the `Stopping container` happening right after the `Running ...` line?

Mymi-INS, 2024-01-31:
I thought the "stopping container" just said that there was an error :') it's not that? (sorry I'm not that good to code)

alejoe91, 2024-01-31:
no, that means that either the spike sorter ran fine, or it failed. In any case, the container in which the spike sorting is run is stopped.

Mymi-INS, 2024-01-31:
How can I know why it stopped? is it for that, that my json is empty and that my code return the error of emptyness?

Also I don't understand why the error message is with the empty json and not with the fact that my container stopped.

alejoe91, 2024-01-31:
no, it should not be empty. Can you share your entire script, and possibly the data?

Mymi-INS, 2024-01-31:
I ll put the script here in 30/40min :) I hope you ll understand something cause there is a lot of stuff in there
For the data is way too heavy, but it's data recorded with openephys (.dat), classical data obtained with silicon probes.

alejoe91, 2024-01-31:
Could you share it with google drive or something similar?

Mymi-INS, 2024-01-31:
Hi again, here is my pipeline, I'll give you a piece of data tomorrow :) Thanks again for your help

```
"""\
Launch the spike sorting algorithm with SpikeInterface.

Usage: in a console: python3 SI_Pipeline.py CONFIG DATE -arg1 ARG1 ... -argn ARGn

Type python3 SI_Pipeline.py -h for help on the possible arguments.
"""

import argparse
import glob
import json
import os
from datetime import datetime
from time import time

import numpy as np
import pandas as pd
import spikeinterface.full as si

def parse_arguments() -> argparse.Namespace:
    """
    Parse CLI arguments.

    Returns:
        Namespace: CLI arguments as an argparse Namespace.
    """
    
    parser = argparse.ArgumentParser()
    
    parser.add_argument("KS config", type=str, help="The name of the config to use.")
    parser.add_argument("date", type=str, help="The date of the recording, used to find the data (eg. 20231212")
    parser.add_argument("-abc", "--additional_bad_channels", type = str, default = "", help = "Additional bad channels. E.g: '17,23,45,69'. (Added to the list specified in the config file).")
    parser.add_argument("--do_correction", type = bool, default = None, help = "Whether drift registration is applied or not.")
    parser.add_argument("-dt", "--detect_threshold", type = float, default = None, help = "Threshold for spike detection.")
    parser.add_argument("--minFR", type = float, default = None, help = "Minimal firing rate required to keep a cluster.")
    parser.add_argument("--minfr_goodchannels", type = float, default = None, help = "Minimal firing rate on a 'good' channel required to keep a cluster.")
    parser.add_argument("-j", "--jobs", type = int, default = 160, help = "Number of subprocesses to use.")
    
    args = parser.parse_args()
    
    for arg in ["jobs", "detect_threshold", "minFR", "minfr_goodchannels"]:
        value = args.__dict__[arg]
        if value is not None:
            assert value >= 0, f"{arg} argument cannot be negative ! (Current value : {value})"
    
    return args

def load_config(args: dict) -> dict:
    """
    Build the config dictionnary from the rat config, the day's base config, the spike sorting config, and the CLI arguments.

    Args:
        args (dict): Input CLI arguments as a Python dict.

    Returns:
        dict: The config dict for this day.
    """

    # Load the spike sorting config
    KS_config_path = os.path.abspath(os.path.join(__file__, os.path.pardir, f"yourpath/KS_config_{args['KS config']}.json"))
    assert os.path.exists(KS_config_path), f"Kilosort config file {KS_config_path} not found. Please check if it exists."
    with open(KS_config_path) as f:
        KS_config = json.load(f)
    
    # Load the rat config
    rat_config_path = os.path.abspath(os.path.join(__file__, f"yourpath/config_{KS_config['rat']}.json"))
    print(rat_config_path)
[... 94 lines cut ...]
        print(f"Sorter output folder writeable: {os.access(KS_output_path, os.W_OK)}")
        print(f"Phy params.py file writeable: {os.access(os.path.join(KS_output_path, 'params.py'), os.W_OK)}")
        

    # Display the running time of the script
    dt = time() - t0
    h = int(dt // 3600)
    m = int((dt - 3600 * h) // 60)
    s = int((dt - 3600 * h - 60 * m))
    
    print(f"Done (in {h}h{m}m{s}s), results stored in {KS_output_path}.")
    
    # Display some stats of the sorting
    labels = pd.read_csv(os.path.join(KS_output_path, "cluster_KSLabel.tsv"), sep="\t")
    n_good = labels["KSLabel"].value_counts()["good"]
    n_mua = labels["KSLabel"].value_counts()["mua"]
    
    print(f"Found {len(labels)} clusters ({n_good} goods, {n_mua} mua).")

if __name__ == "__main__":
    args = parse_arguments()
    config = load_config(args.__dict__)
    run_cluster_cutting(config)

```

Mymi-INS, 2024-02-01:
Hi Alessio,
I send you an email with the link for the data :)
Thanks again for your help !

zm711, 2026-10-30:
I will close this as stale at this point, but if more issues come up just open a new issue.

closed 2026-10-30 by zm711