SpikeInterface/spikeinterface #2461: Question about waveforms templates

Antho2422, 2024-02-02:
Hi all, 

I have a question about the waveforms templates and metrics, it is pretty messy in my head right now and I would like to clarify some points. 

It's a bit tricky because I'm not working on my own data but I am working on data from somebody else in the lab so I'm not 100% aware of all the exact steps he did before but I am importing directly the data from his Phy folder. So I will try to summarize as clearly as possible my goal and work flow. 

### **Goal :**
Load all sortings already done outside of spike interface a few month ago using (SC1) from the **phy folders**, extract the waveforms and then compute **metrics on each individual cluster templates.** The final goal is to try to develop a method to help with the curation, so we also have for each cluster the label assigned in Phy by my colleague.

### Steps : 

- Load the sorting from the phy folder :
```python
sorting = se.read_phy(phy_path, exclude_cluster_groups=[])
``` 
- Load the associated recording : 
```python
recording = se.read_neuralynx(folder_path=session_path, stream_id=None, exclude_filename=to_exclude)
``` 
- Channel slice the recording according to the channel that were selected by my colleague : 
```python
recording = recording.channel_slice(channel_ids=ids[session_path])
``` 

At this point I should have the same data he used to do the sorting + the sorting object imported after curation. 

- I create a probe with random location because it does not have importance in our case since we don't know the exact position of our contacts since we are working on Human data with a few wires 
```python
    linear_probe = generate_linear_probe(num_elec=len(recording.get_channel_ids()), ypitch=50)
    indices = np.arange(len(recording.get_channel_ids()))
    linear_probe.set_device_channel_indices(indices)
    recording = recording.set_probe(linear_probe)
    sorting.register_recording(recording=recording)
``` 
- Now I extract the waveforms (please note that I have tried with both sparse=True and sparse=False):
```python
    waveforms = si.extract_waveforms(folder=os.path.join(wf_path, patient), recording=recording_filtered, 
                                     sorting=sorting, sparse=False, ms_before=2, ms_after=2, max_spikes_per_unit=None, 
                                     seed=42, dtype='float64', load_if_exists=True)
``` 

- My final step is to create a dataframe with metrics about each cluster + some metadata using this process : 
```python
def get_data_cluster(recording, waveform, sorting, patient, session):
    print(f'Computing metrics for patient {patient}')
    if waveform.sorting is None:
        waveform.sorting = sorting
    if waveform.get_sorting_property('quality') is None:
        raise ValueError(f'No quality for patient {patient}')
    _ = spost.compute_spike_amplitudes(waveform, load_if_exists=False)
    _ = spost.compute_template_metrics(waveform, load_if_exists=False)
    _ = spost.compute_principal_components(waveform, n_components=2, load_if_exists=False)
    metrics = compute_quality_metrics(waveform_extractor=waveform, load_if_exists=False)
    template_metrics = waveform.load_extension('template_metrics')
    df = template_metrics.get_data()
    print("Number of units : ", len(df))
    df = pd.concat([df, metrics], axis=1)
    df['wf_score'] = [compute_wvscore(waveform.get_waveforms(unit_id=id)) for id in waveform.unit_ids]
    df['amp'] = waveform.get_sorting_property('amp')
    df['channel'] = waveform.get_sorting_property('ch')
    df['n_spikes'] = waveform.get_sorting_property('n_spikes')
    df['label'] = waveform.get_sorting_property('quality')
    df['ori_id'] = df.index
    df['patient'] = [patient] * len(df)
    df['session'] = [session] * len(df)
    df['duration'] = [recording.get_total_duration()] * len(df)
    df = df.sort_values(by='amp', ascending=False)
    return df
``` 

**After that I have one data frame for each of my patient and then I merge every dataframe into one dataframe with all cluster from every patient.** 

### The issue :
After this process I am trying to verify visually some cluster to make sure the label was right ("good", "mua" or "noise). So I decided to plot some of the templates.I load the waveforms using the load_waveforms() function from the waveforms folder created when extracting them. This is what I expect for most of the "good", plus I am also expecting the metrics to be good aswell. 

[image]

But in some cases I got very weird templates and weird waveforms in general on units classified as "good" units. And this is not only coming from the visualisation because I also get weird metrics from the template. In this specific case also it seems like it does not even plot all the waveforms (even if I ask for 500 waveforms in the plotting function) and the cluster has more than 3000 spikes :

[image]

Here is my plotting function also in case I'm doing something I should not do : 
```python
def plot(waveforms, unit_id, patient, channel_idx, n_spikes, df_index):
    channel_idx = get_template_extremum_channel(waveforms, outputs='index', peak_sign="pos")[unit_id]
    fig, ax = plt.subplots(figsize=(10, 10))
    template = waveforms.get_template(unit_id, mode="average")[:, channel_idx]
    std = waveforms.get_template(unit_id, mode='std')[:, channel_idx]
    random_spikes = np.random.randint(0, waveforms.get_waveforms(unit_id).shape[0], n_spikes)
    for i in (random_spikes):
        ax.plot(
            waveforms.get_waveforms(unit_id)[i, :, channel_idx], color='k', alpha=0.09,
                )
        ax.plot(
            template, color='r', linewidth=2,
        )
        ax.fill_between(
            np.arange(template.shape[0]), template - std, template + std, color='r', alpha=0.2, zorder=1
        )
    ax.set_title(f'Unit {unit_id} on channel {channel_idx} patient {patient} (index {df_index})')
    ax.set_ylabel('Amplitude (uV)')
    ax.set_xticks([])
    plt.show()
``` 

### Summary :

I don't know where the issue could come from :

- Wrong classification from my colleague (classified some units as good when it was noise) but I doubt that because it happens a lot almost a third of the good units. 
- Problem related to the recording
- Issue during waveforms extraction
- My code is wrong or I'm missing some steps
- Something else

If anyone can help with that I would be really grateful, I can also provide some test data in case you need, but the data is very large so I'm not sharing it for the moment !

Thank you, 
Anthony

alejoe91, 2024-02-05:
Hi @Antho2422 

The first image you shared is not displaying. Can you paste it again?

Why are you using `peak_sign="pos"` when extracting templates? The waveforms that you are plotting seem `neg`

Antho2422, 2024-02-05:
For this specific unit from the picture above the waveforms looks good now somehow... I must have made a mistake sorry.

But now I'm having the issue on another unit, the issue is probably coming from my code or from the way I extract the waveforms... 

Here is another unit that has the following metrics : 

```python
peak_to_valley : 0.00103759765625
peak_trough_ratio : -0.29893273
half_width : 0.000299072265625
repolarization_slope : 77.6452599942881
recovery_slope : -12.54579060918648
num_positive_peaks : 2
num_negative_peaks : 1
num_spikes : 7917.0
firing_rate : 1.0995836828815724
presence_ratio : 0.957983193277311
snr : 0.0072501116
isi_violations_ratio : 0.650936665932756
isi_violations_count : 17.0
rp_contamination : 0.5510514284838923
rp_violations : 7.0
sliding_rp_violation : nan
amplitude_cutoff : 0.0222095143422889
amplitude_median : 51.999305725097656
amplitude_cv_median : nan
amplitude_cv_range : nan
sync_spike_2 : 0.0207149172666414
sync_spike_4 : 0.000126310471138
sync_spike_8 : 0.0
firing_range : 2.8
isolation_distance : 13778.187104592098
l_ratio : nan
d_prime : 0.0395080266906735
silhouette : -0.9992417106908028
nn_hit_rate : 1.0
nn_miss_rate : 0.9997208971886736
wf_score : 0.0111489904474519
amp : 66.36216736
channel : 4
n_spikes : 7917
label : good
``` 

And when I plot all the waveforms (all 7917 spikes) it has weird shape of template (same using the median instead of average) : 

[image]

By the way, this is another unit looking good : 

[image]

Antho2422, 2024-02-05:
I noticed that in the metrics the amplitude of the unit that has a weird shape is peaking at 66 and this is what we see visually (almost) so I don't understand why the template is so low...

alejoe91, 2024-02-05:
looking at the second unit, it seems that the code is doing it's job. For the first unit, probably there are many small/zero waveforms. Can you plot all waveforms without the template and the template shading?

Antho2422, 2024-02-05:
Of course here it is. 
[image]

About the small/zero waveforms, I do not really understand why the templates metrics extract an amplitude of 66 if the template is close from 0. I extracted the waveforms with `sparse=False`. In my case I have waveforms most of the times picked up on one channel only so I do not care about the other channel infos. I'm using the `channel_idx = get_template_extremum_channel(waveforms, outputs='index', peak_sign="pos")[unit_id]` to get the channel index but I don't know if it is necessary when having not sparse waveforms.

alejoe91, 2024-02-05:
yeah that's weird. Most waveforms are 0 right? What is the black think line?

How are you getting the metrics for that unit? Maybe something is wrong when indexing the dataframe...

Antho2422, 2024-02-05:
Yes, most of the waveforms are 0 or close. Maybe the indexing is wrong but I double checked it and it seems okay. I will have another look.

Antho2422, 2024-02-05:
@alejoe91 

Okay I think I have a clue of what is happening, in some cases, the data seems to be in V instead of mV. I don't know exactly why but my template amplitudes seems very very low as you can see here : 
[image]

alejoe91, 2024-02-05:
mmm, maybe the scaling in the original file is not correct. Note that amplitudes after scaling should be in uV, not mV!

Antho2422, 2024-02-05:
Yes my bad I meant uV. I will try to investigate a bit more and keep you updated if I have more information on what could be the issue here

Antho2422, 2024-02-06:
@alejoe91 

I think I figured it out. There was a mismatch between the number of channels in my recording, and the number of channels the sorting has been performed on. So it was extracting the template on another channel and the metrics were very bad. 

Thank you for your help :)

Cheers

Antho2422, 2024-02-06:
@alejoe91 sorry to re-open it but I noticed a weird behavior that can be related to that. 
```python
print(f'Number of channels : {recording.get_num_channels()}')
print(f'Expected number of channels based_on waveforms : {waveforms.get_waveforms(sorting.get_unit_ids()[0]).shape[2]}')
``` 

This code outputs me a different number of channels in the recording vs waveforms. I am somehow "loosing" one channel information during the waveforms extraction.  For example, the above code returns 4 and 3 I'm not sure why and this causes errors when running the `get_template_extremum_channel()` function that sometimes returns me channel indexes out of range...

alejoe91, 2024-02-06:
If sparse is false then you should get all channels...

Antho2422, 2024-02-06:
Sparse is `True`

alejoe91, 2024-02-06:
Oh, then you should set it to False if you want the waveforms on all channels!

Antho2422, 2024-02-06:
I'm not sure to understand the sparse arg because when I set it to True I have access to data from 3 channels  still

alejoe91, 2024-02-06:
to extract waveforms on all channels, you have to use `sparse=False`. If `sparse=True`, it will use channel locations to only select channels in the vicinity of the waveform peak

Antho2422, 2024-02-06:
okay I get it now, thanks a lot !!