SpikeInterface/spikeinterface #2463: custom probe design

kshtjkumar, 2024-02-04:
hi, I am trying to design a 15 ch probe, using this:
```
from probeinterface import generate_multi_columns_probe
square_probe = generate_multi_columns_probe(num_columns=4,
                                            num_contact_per_column=4,
                                            xpitch=5.2, ypitch=5,
                                            contact_shapes='circle', contact_shape_params={'radius': 2})

square_probe.create_auto_shape('rect')
plot_probe(square_probe)
plt.show()
```

[image]

is it possible to remove the last channel (4th row last ch), so that the total electrodes are 15 ?

samuelgarcia, 2024-02-05:
`generate_multi_columns_probe()`is a fast generator.
If you want more control you can create a probe from sractch using this kind of code  : 
https://probeinterface.readthedocs.io/en/main/examples/ex_01_generate_probe_from_sratch.html#sphx-glr-examples-ex-01-generate-probe-from-sratch-py

In short you create a positions array of shape (15, 2).

The other possibility is cerate manually an excel file and read it as a dataframe and then do
`probe = Probe.from_dataframe(pd.read_excel('/my/probe.xlsx))`

alejoe91, 2024-02-05:
@kshtjkumar 

You can also do the following:
```
from probeinterface import probe

probe_array = square_probe.to_numpy(complete=True)

# skip last contact here
probe_without_last_channel = Probe.from_numpy(probe_array[:-1])
```