SpikeInterface/spikeinterface #2467: Windows fatal exception when using run_sorter_by_property

Ashkees, 2024-02-05:
Hello,

I wrote a little script to cluster my data and it works fine on my computer.  Now I'm trying to run the same on the computer of my colleague and after I run this code:
```
sorting_SC2 = si.run_sorter_by_property('spykingcircus2', recording, grouping_property='group',
                                            working_folder=base_folder / 'results_SC2',
                                            verbose=True, job_kwargs=job_kwargs)
```
I get the following output and error message:
```
detect peaks using locally_exclusive with n_jobs = 1 and chunk_size = 20000
detect peaks using locally_exclusive:   0%|          | 0/1723 [00:00<?, ?it/s]
We found 715140 peaks in total

Windows fatal exception: code 0xc06d007e

Thread 0x00000f88 (most recent call first):
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\threading.py", line 324 in wait
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\threading.py", line 607 in wait
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\tqdm\_monitor.py", line 60 in run
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\threading.py", line 1016 in _bootstrap_inner
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\threading.py", line 973 in _bootstrap

Main thread:
Current thread 0x000047d0 (most recent call first):
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\scipy\linalg\_basic.py", line 1282 in lstsq
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\scipy\signal\_savitzky_golay.py", line 142 in savgol_coeffs
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\scipy\signal\_savitzky_golay.py", line 341 in savgol_filter
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sortingcomponents\clustering\random_projections.py", line 119 in main_function
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sortingcomponents\clustering\main.py", line 42 in find_cluster_from_peaks
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\internal\spyking_circus2.py", line 112 in _run_from_folder
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py", line 254 in run_from_folder
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py", line 173 in run_sorter_local
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py", line 147 in run_sorter
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\launcher.py", line 102 in run_sorter_jobs
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\launcher.py", line 293 in run_sorter_by_property
  File "c:\users\aerhardt\documents\anael\syt7_bootcamp\si_clustering_scripts\main.py", line 35 in load_sort_save
  File "c:\users\aerhardt\documents\anael\syt7_bootcamp\si_clustering_scripts\main.py", line 61 in <module>
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spyder_kernels\py3compat.py", line 356 in compat_exec
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 473 in exec_code
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 615 in _exec_file
  File "C:\Users\aerhardt\AppData\Local\anaconda3\envs\si_env\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 528 in runfile
  File "C:\Users\aerhardt\AppData\Local\Temp\ipykernel_16516\4121288921.py", line 1 in <module>

Restarting kernel...
```
Any advice for dealing with this issue is much appreciated!
Thanks in advance,
Ashley

Ashkees, 2024-02-05:
Ok, so maybe I'm making a silly error that has something to do with python that I don't understand yet.  When I run on my computer, I run it using PyCharm and it works fine.  On my colleague's computer it works ok if I run it from the Anaconda console, but not if I run it using Spyder (this is where I was getting the above errors).  So, maybe everything is ok with spikeinterface and it's more of a matter of me not knowing how to run my scripts properly.

alejoe91, 2024-02-05:
Hi @Ashkees 

Yeah I think that Spyder might be more complicated in terms of properly controlling environments resources. Should we close the issue then?

zm711, 2024-02-05:
This type of error on Windows is usually do to environment stuff getting out of sync. It might be best for you to just create a fresh environment. Since you created this with the installation tips yaml, how did you sync up spyder? Are you using (base) spyder, but with a kernel from your si_env? 

I'm curious since I've been able to coordinate Spyder with spikeinterface, so I don't think this should be impossible. (based on my reading your error `Windows fatal exception: code 0xc06d007e` typically refers to corrupted files on Windows), so a fresh environment would like get rid of whatever file got corrupted between Spyder and the other files in your env.

I'm happy to explore this a little more @alejoe91 and close if we can't figure this out. I think we should be able to get SI working everywhere : )

samuelgarcia, 2024-02-05:
Hi Ashley.
Could you try to run the same in a conda consol in windows instead of spyder ?
Just to check that it is windows related and not ipykernel embeded in spyder related.

Ashkees, 2024-02-06:
> This type of error on Windows is usually due to environment stuff getting out of sync. It might be best for you to just create a fresh environment. Since you created this with the installation tips yaml, how did you sync up spyder? Are you using (base) spyder, but with a kernel from your si_env?

I created the si_env using the installation tips yaml.  Then I ran 'conda install spyder' in that environment, enabling me to run spyder from the si_env.  

> I'm curious since I've been able to coordinate Spyder with spikeinterface, so I don't think this should be impossible. (based on my reading your error `Windows fatal exception: code 0xc06d007e` [...]

It occurs to me now that I should have added spyder to the yaml and created a fresh environment all at once, instead of adding in programs after.  I don't have access to my colleague's computer today, but when I do, I'll give that a try and anything else you suggest =)

zm711, 2024-02-06:
> It occurs to me now that I should have added spyder to the yaml and created a fresh environment all at once, instead of adding in programs after. I don't have access to my colleague's computer today, but when I do, I'll give that a try and anything else you suggest =)

I think that is the best case scenario for a potentially storage expensive solution. Spyder is such a complicated program with so many dependencies it is probably best to let the conda resolver take care of the hard dependency resolution if you want Spyder in your env. I think Sam's suggestion could also work playing around with the ipykernel you use ([docs here for Spyder](https://docs.spyder-ide.org/current/panes/ipythonconsole.html)). 

But I don't think it is a good idea to `pip install Spyder` after creating the env. Either put it at the beginning (in the yaml) or use a Spyder-ipykernel strategy.

zm711, 2024-03-08:
@Ashkees, any update on getting things to work for you?

Ashkees, 2024-03-14:
Hi, thanks for checking in.  Putting Spyder in the .yml and creating the environment from that did not solve the issue.  I haven't had a chance to try the Spyder-ipykernel strategy.  For my purposes, I've had to move on from that aspect of the project, so we can close the issue for now.  I think when I come back to it I'll forget about Spyder and run my spike interface scripts from PyCharm or from the Anaconda command prompt instead.

Thanks for your help!  Sorry I wasn't able to follow through with all your suggestions!