SpikeInterface/spikeinterface #2469: Bug in template.py since zarr handling

yger, 2024-02-05:
Line 210 and 277 in template.py are triggering errors on my machine, since the last push of zarr handling for Templates. The error is:

TypeError: unsupported operand type(s) for |: 'type' and 'type'

It can be removed if I remove the typing in the signatures of the function, since this is the str | Path that is problematic

h-mayorquin, 2024-02-05:
This is most likely python 3.8 and 3.9 related. This should be solved after:
https://github.com/SpikeInterface/spikeinterface/pull/2468

zm711, 2024-02-06:
I'll close this since with #2468 there is `from __future__ import annotations` everywhere now which should prevent any typing issues. Reopen if the issue wasn't solved.

closed 2024-02-06 by zm711