SpikeInterface/spikeinterface #2470: ValueError: The given Probe have 'device_channel_indices' that do not match channel count  63 vs 60

NathanielH-snek, 2024-02-05:
Greetings. I have a batch script to process multiple recordings and I am currently running into an error. Each recording is a tetrode recording with either 62,63,64, or 67 channels. As such I need to cut out any channels that are labeled Auxillary and then also remove an extra channels that are not part of a full tetrode. In my following code using it on a recording that is originally 62 channels I get the error listed in the title. When looking at the code the prompts the error there is a TODO comment above it. I am wondering if there is some sort of extrapolation to add channels to the tetrode maps because the length of the channel_nums list I have is 60 and exporting the probe to a dataframe to look at all the information points towards no errors there either. Perhaps I am simply making an error in my script somewhere but this may also be something inherent to spikeinterface itself currently. Thank you for your help
```

origChanIDs = recordingObject.get_channel_ids().tolist()
auxChannels = ['AUX1','AUX2','AUX3','aux1','aux2','aux3']
newChanIDs = [channel for channel in origChanIDs if channel not in auxChannels]
numChannels = len(newChanIDs)
totalTetrodeChannels = int(4 * math.floor(numChannels/4))
channelsToCut = newChanIDs[:numChannels-totalTetrodeChannels]
x = [0] * numChannels
y = list(range(1, numChannels+1))
locations = list(zip(x,y))
locations = np.array(locations)
locations.shape
newChanIDs = np.asarray(newChanIDs)
recordingObject.set_channel_locations(locations,newChanIDs)
recordingObject = recordingObject.remove_channels(channelsToCut)
print(recordingObject)
print(recordingObject.get_channel_ids())
import re
channel_nums = recordingObject.get_channel_ids()
num_channels = recordingObject.get_num_channels()
channel_nums = channel_nums[0:totalTetrodeChannels].tolist()
for i in range(len(channel_nums)):
    word = str(channel_nums[i])
    word = re.findall(r'\d+' , word)
    channel_nums[i] = int(word[0]) - 1
        

probeGroup = ProbeGroup()
for i in range(int(totalTetrodeChannels/4)):
    tetrode = generate_tetrode()
    tetrode.move([i * 50, 0])
    probeGroup.add_probe(tetrode)
probeGroup.set_global_device_channel_indices(channel_nums)

print(channel_nums)

recordingWProbe = recordingObject.set_probegroup(probeGroup,group_mode="by_probe")

```

alejoe91, 2024-02-05:
Hi @NathanielH-snek 

The `device_channel_indices` indicate the channel index that each contact correspond to. Therefore, the highest device channel index CANNOT exceed the number of channels - 1, because it indexes the traces.

NathanielH-snek, 2024-02-05:
So this is not a result of the actual number of the labels but rather the actual values associated with them if I'm understanding this correctly?

closed 2024-02-24 by NathanielH-snek