SpikeInterface/spikeinterface #2472: UFuncTypeError: Cannot cast ufunc 'subtract' output from dtype('float64') to dtype('int64') with casting rule 'same_kind'

chengtang827, 2024-02-05:
I got the following error when trying to do motion correction.
```
---------------------------------------------------------------------------
UFuncTypeError                            Traceback (most recent call last)
Cell In[13], line 2
      1 motion_folder = Path("motion_folder")
----> 2 recording_motion = spre.correct_motion(recording_processed,
      3                                        preset=preset,
      4                                        folder=motion_folder,
      5                                        progress_bar=True,
      6                                        estimate_motion_kwargs=optimized_estimate_motion_kwargs)
      7 motion_info = spre.load_motion_info(motion_folder)

File /net/oms.ib.cluster/om4/group/jazlab/apiccato/spikeinterface/src/spikeinterface/preprocessing/motion.py:331, in correct_motion(recording, preset, folder, output_motion_info, detect_kwargs, select_kwargs, localize_peaks_kwargs, estimate_motion_kwargs, interpolate_motion_kwargs, **job_kwargs)
    328     np.save(folder / "peak_locations.npy", peak_locations)
    330 t0 = time.perf_counter()
--> 331 motion, temporal_bins, spatial_bins = estimate_motion(recording, peaks, peak_locations, **estimate_motion_kwargs)
    332 t1 = time.perf_counter()
    333 run_times["estimate_motion"] = t1 - t0

File /net/oms.ib.cluster/om4/group/jazlab/apiccato/spikeinterface/src/spikeinterface/sortingcomponents/motion_estimation.py:150, in estimate_motion(recording, peaks, peak_locations, direction, bin_duration_s, bin_um, margin_um, rigid, win_shape, win_step_um, win_sigma_um, post_clean, speed_threshold, sigma_smooth_s, method, output_extra_check, progress_bar, upsample_to_histogram_bin, verbose, **method_kwargs)
    148 # run method
    149 method_class = estimate_motion_methods[method]
--> 150 motion, temporal_bins = method_class.run(
    151     recording,
    152     peaks,
    153     peak_locations,
...
--> 330     bins -= np.mean(bins)
    331     smooth_kernel = np.exp(-(bins**2) / (2 * histogram_time_smooth_s**2))
    332     smooth_kernel /= np.sum(smooth_kernel)

UFuncTypeError: Cannot cast ufunc 'subtract' output from dtype('float64') to dtype('int64') with casting rule 'same_kind'
```
For now I got rid of the error by replacing `line 330` into: 
`bins = bins - np.mean(bins)`
But I guess there should be more principled ways to avoid the type casting error?

alejoe91, 2024-02-06:
Hi @chengtang827 

This is fixed in main and in `0.100.0` that we just released this morning :)
Can you upgrade and run it again?

alejoe91, 2024-02-08:
Fixed in 0.100.0