SpikeInterface/spikeinterface #2474: Assigning memory usage to sorters

m-bellv, 2024-02-07:
Hello,

I am trying to run several spike sorters using spikeinterface. I assign the number of jobs and memory using 
global_job_kwargs = dict(n_jobs=12, total_memory='180G') 
si.set_global_job_kwargs(**global_job_kwargs)

When I try to run spikesorting using spykingcircus2, using 
sorting_spykingcircus2 = ss.run_sorter(sorter_name="spykingcircus2", recording=rec_w,
                                       output_folder="2021-11-29/spkinterface/spykingcircus2",
                                       filtering={},
                                       apply_preprocessing = False,
                                       job_kwargs = {'n_jobs': 12, 'total_memory':'180G'},
                                       detection = {'peak_sign': 'neg', 'detect_threshold': 6}, 
                                       selection = {"n_peaks_per_channel": 10000,
                                                    "min_n_peaks": 20000},
                                       )
I get the error: numpy.core._exceptions._ArrayMemoryError: Unable to allocate 392. GiB for an array with shape (5, 231, 91107520) and data type float32

I tried including matching = {"method": "circus-omp-svd"} within the sorter arguments, but the error persists.
Is there a way to restrict the memory used by spykingcircus or adjust the chunk size? I am using spikeinterface version: 0.99.1.

m-bellv, 2024-02-07:
I am also having errors with pykilosort, where I get the error 
pykilosort-1.5.0-py3.10.egg/pykilosort/learn.py", line 69, in extractTemplatesfromSnippets
    wTEMP[:, j] = cp.dot(dd[:, imax == j], amax[imax == j].T)
  File "cupy/_core/core.pyx", line 1592, in cupy._core.core._ndarray_base.__setitem__
  File "cupy/_core/_routines_indexing.pyx", line 50, in cupy._core._routines_indexing._ndarray_setitem
  File "cupy/_core/_routines_indexing.pyx", line 996, in cupy._core._routines_indexing._scatter_op
  File "cupy/_core/_routines_indexing.pyx", line 409, in cupy._core._routines_indexing._view_getitem
IndexError: Index 0 is out of bounds for axis 1 with size 0

Even though I have set do_whitening=False when running the pykilosort sorter: 
sorting_pykilosort = ss.run_sorter(sorter_name="pykilosort", recording=rec_w,
                                   output_folder="/2021-11-29/spkinterface/pykilosort",
                                   do_whitening=False,
                                   n_jobs = 12)

Ironclust, mountainsort4, mountainsort5, tridesclous, and waveclus seem to run fine without errors.

yger, 2024-02-07:
Something is weird here, since spykingcircus 2 should not allocate such an amount of ram. I guess there is something weird with your preprocessing steps. What are they ?

m-bellv, 2024-02-07:
from probeinterface.plotting import plot_probe
from probeinterface import Probe
import spikeinterface.core as si
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.curation as scur
import spikeinterface.widgets as sw
import numpy as np
import pandas as pd

ncore =12
global_job_kwargs = dict(n_jobs=ncore, total_memory='180G') 
si.set_global_job_kwargs(**global_job_kwargs)

# NumpyRecordingExtractor
Fs = 24414.0625
num_channels = 16
channelIDs = np.array(range(num_channels))

# Create probe
n = 16
positions = np.zeros((n, 2))
for i in range(n):
    x = i // 4
    y = i % 4
    positions[i] = x, y
positions *= 500

probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=positions)
probe.set_device_channel_indices(channelIDs)

# Load data convert to uV from V
data = np.load('/Pr1/2021-11-29/MUAfiltdata_002.npy',
               allow_pickle=True).T*1000000

# import data to spike interface
recording = se.NumpyRecording(
    data, sampling_frequency=Fs, channel_ids=channelIDs)

# # create dummy stimulation triggers per segment
stimulation_trigger_frames = np.array([1468970, 78885536])
# Remove artifact periods from recording
recording_rm_artifact = spre.remove_artifacts(
    recording, stimulation_trigger_frames, ms_before=10, ms_after=6110)

rec_w = spre.whiten(recording=recording_rm_artifact)

# Assign probe file to recordings
rec_w = rec_w.set_probe(probe)
rec_w.annotate(is_filtered=True)

rec_w = rec_w.save(folder='/Pr1/2021-11-29/spkinterface/data', format='binary')

yger, 2024-02-08:
Is the file very large, and/or can you share it with me? I don't get at all this bug, this does not make sense to me and thus I have no idea where does it comes from. Do you have the error message immediatly at the beginning of SC2 (if you have verbose=True)? Can you share more of the error message/console output?

m-bellv, 2024-02-13:
The file is very large, so I expect it would be difficult to share. I could parse it but I believe the error is a result of the file size, as it seems to be including the whole recording when trying to analyze. I believe the error has to do with setting the job_kwargs = dict(n_jobs=12, total_memory='180G'), as when Submitting: 

sorting_spykingcircus2 = ss.run_sorter(sorter_name="spykingcircus2", recording=rec_w,
...                                        output_folder="/Pr1/2021-11-29/spkinterface/spykingcircus2",
...                                        filtering={},
...                                        apply_preprocessing = False,
...                                        job_kwargs = {'n_jobs': ncore, 'total_memory':'180G'},
...                                        detection = {'peak_sign': 'neg', 'detect_threshold': 6}, 
...                                        selection = {"n_peaks_per_channel": 10000,
...                                                     "min_n_peaks": 20000},
...                                        verbose = True)

The full message is: 

detect peaks using locally_exclusive with n_jobs = 12 and chunk_size = 234375000
detect peaks using locally_exclusive: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:17<00:00, 17.51s/it]
We found 150872 peaks in total
We kept 150872 peaks for clustering
extracting features with n_jobs = 12 and chunk_size = 234375000
extracting features: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:09<00:00,  9.68s/it]
We found 477 raw clusters, starting to clean with matching...
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:07<00:00,  7.84s/it]
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:07<00:00,  7.95s/it]
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:07<00:00,  7.93s/it]
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:08<00:00,  8.09s/it]
We kept 303 non-duplicated clusters...
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:07<00:00,  7.88s/it]
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:07<00:00,  7.86s/it]
extract waveforms shared_memory multi buffer with n_jobs = 12 and chunk_size = 234375000
extract waveforms shared_memory multi buffer: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:08<00:00,  8.02s/it]
find spikes (circus-omp-svd) with n_jobs = 12 and chunk_size = 234375000
find spikes (circus-omp-svd):   0%|                                                                                                                           | 0/1 [00:08<?, ?it/s]
Error running spykingcircus2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 147, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 173, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 289, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
concurrent.futures.process._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/python/3.10.2/lib/python3.10/concurrent/futures/process.py", line 243, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/python/3.10.2/lib/python3.10/concurrent/futures/process.py", line 202, in _process_chunk
    return [fn(*args) for args in chunk]
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/python/3.10.2/lib/python3.10/concurrent/futures/process.py", line 202, in <listcomp>
    return [fn(*args) for args in chunk]
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 437, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
[... cut ...]
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/sortingcomponents/peak_detection.py", line 117, in detect_peaks
    outs = run_node_pipeline(
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/core/node_pipeline.py", line 459, in run_node_pipeline
    processor = ChunkRecordingExecutor(
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 348, in __init__
    self.chunk_size = ensure_chunk_size(
  File "/Pr1/env2/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 211, in ensure_chunk_size
    assert total_memory is None
AssertionError

yger, 2024-02-13:
Can you show me a plot of the traces? Because I think there is something weird going on. The error seems to be related to the fact that circus is finding way too many templates. Since this does not make sense, the question is "are the data properly loaded" and/or why is that so... you said other sorters like tridesclous  or mountain sort worked ?

m-bellv, 2024-02-13:
Yes, Ironclust, mountainsort4, mountainsort5, tridesclous, and waveclus seem to run fine without errors. spykingcircus2, pykilosort, tridesclous2, herdingspikes, and hdsort fail. Perhaps it is worth mentioning that I am able to run the same dataset  using a spykingcircus gui (directly loading the .npy file) installed to my desktop outside of spikeinterface without errors, so I am not sure the issue is with the data. 
Here I have plotted the first 30s using: 
plt.plot(rec_w.get_traces()[0:int(30*Fs),0])
The y-axis is in uV. This is after the rec_w = spre.whiten(recording=recording_rm_artifact) preprocessing as described above.
The dataset has dimensions (91107328, 16), so 16 channels and 91107328 samples per channel.

[image]

m-bellv, 2024-02-13:
I actually seem to be able to get tridesclous2 and pykilosort to work by setting n_jobs to 1,

sorting_TDC2 = ss.run_sorter(sorter_name="tridesclous2", recording=rec_w, 
                            output_folder="/Pr1/2021-11-29/spkinterface/tridesclous2",
                            apply_preprocessing = False,
                            filtering = {},
                            job_kwargs = dict(n_jobs=1, total_memory='180G'),
                            verbose = True
                            )

sorting_pykilosort = ss.run_sorter(sorter_name="pykilosort", recording=rec_w,
                                   output_folder="/Pr1/2021-11-29/spkinterface/pykilosort",
                                   do_whitening=False,
                                   n_jobs = 1,
                                   verbose = True)

 I was previously getting a similar error for both as with spykingcircus2 when I try to adjust the chunksize: 
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending. 
So I do think the issue I am having has to do with the parallelization, although it is interesting that for ironclust, mountainsort4, and tridesclous I am able to run successfully when inputing n_jobs = 12. 

When I restrict
ncore = 1
global_job_kwargs = dict(n_jobs=ncore, chunk_size = int(Fs))
si.set_global_job_kwargs(**global_job_kwargs)

And then submit sorting_spykingcircus2 = ss.run_sorter(sorter_name="spykingcircus2", recording=rec_w,
                                       output_folder="/home/bellvila/projects/def-jmclauri/bellvila/Pr1/2021-11-29/spkinterface/spykingcircus2",
                                       filtering={},
                                       apply_preprocessing = False,
                                       job_kwargs = {'n_jobs': 1, 'chunk_size' : int(Fs), 'total_memory':'500M'},
                                       detection = {'peak_sign': 'neg', 'detect_threshold': 6}, 
                                       selection = {"n_peaks_per_channel": 10000,
                                                    "min_n_peaks": 20000},
                                       verbose = True)

It still seems to crash unfortunately but without giving any errors:
detect peaks using locally_exclusive with n_jobs = 1 and chunk_size = 24414
detect peaks using locally_exclusive: 100%|#############################################################################################| 3732/3732 [00:12<00:00, 309.26it/s]
We found 150872 peaks in total
We kept 150872 peaks for clustering
extracting features with n_jobs = 1 and chunk_size = 24414
extracting features: 100%|##############################################################################################################| 3732/3732 [00:08<00:00, 427.68it/s]
We found 477 raw clusters, starting to clean with matching...
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|####################################################################################| 3732/3732 [00:02<00:00, 1392.99it/s]
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|####################################################################################| 3732/3732 [00:02<00:00, 1432.99it/s]
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|####################################################################################| 3732/3732 [00:00<00:00, 5991.09it/s]
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|#####################################################################################| 3732/3732 [00:08<00:00, 447.59it/s]
We kept 303 non-duplicated clusters...
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|####################################################################################| 3732/3732 [00:03<00:00, 1119.19it/s]
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|####################################################################################| 3732/3732 [00:01<00:00, 2444.57it/s]
extract waveforms shared_memory multi buffer with n_jobs = 1 and chunk_size = 24414
extract waveforms shared_memory multi buffer: 100%|#####################################################################################| 3732/3732 [00:06<00:00, 585.51it/s]
find spikes (circus-omp-svd) with n_jobs = 1 and chunk_size = 24414
find spikes (circus-omp-svd):   0%|    

And just freezes there

yger, 2024-02-13:
I think I've found the problem ! Just to be sure, the problem does not happen if you remove the total_memory argument?

m-bellv, 2024-02-13:
A problem occurs even without the total_memory argument if n_jobs is set to anything greater than 1 for spykingcircus2, tridesclous2, and pykilosort. I get the error: 
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending

Although the memory problem is resolved without including it, yes.

yger, 2024-02-14:
Ok. So the problem for the memory usage for SC2 has been fixed in the dev branch (for now on, the PR are frozen due to major refactoring). Regarding the n_job issues, have you tried updating threadpool?

yger, 2024-02-14:
ping for @samuelgarcia not to forget: the problem is that fix_job_kwargs do not allow overwritting of choices for memory usage. It has to be better handled, otherwise we can not overwrite global settings

m-bellv, 2024-02-14:
> Ok. So the problem for the memory usage for SC2 has been fixed in the dev branch (for now on, the PR are frozen due to major refactoring). Regarding the n_job issues, have you tried updating threadpool?

Yes, my threadpoolctl is version 3.2.0

m-bellv, 2024-02-21:
I seem to be able to run with multiple jobs for all sorters if I set mp_context = 'spawn' (it fails when set to None or 'fork'), even though I am running spikeinterface on a linux server.