SpikeInterface/spikeinterface #2476: Retrieve unmerged clusters when reading Phy

Antho2422, 2024-02-07:
Hi, 

I was wondering if there exist any way to retrieve the spike times and waveforms of unmerged clusters after they were merged in Phy ? And if not, is there a way to retrieve the information of how many original clusters were merged in a new cluster if there was a merge ?

Thank you, 
Anthony

alejoe91, 2024-02-07:
Hi Anthony,

Unfortunately, I don't think there is a way. Can you ask on the Phy repo?

zm711, 2024-02-07:
I'm running the issues on the Phy repo now. And as far as I know the answer is no. Sorry Anthony. You could try playing with the cache files, in the `.phy`, but phy rewrites over all the `.npy` files once you hit save for a merge, so it gets overwritten for everything.

alejoe91, 2024-02-07:
Another option is the following:
After exporting to phy, copy all tsv files to a new folder.

To restore the Phy original output (pre-curation), delete all tsv files and the `spike_clusters.npy` and move the original ones to the Phy folder.

zm711, 2024-02-07:
One of my labmates goes a step farther. Just to be safe she copies the whole folder at multiple stages to keep full records of everything at each stage. It's very storage expensive. :)

Antho2422, 2024-02-07:
Okay thank you for all your answers. I will try to do that but yes in fact it is very storage expensive.

zm711, 2024-02-07:
Not to toot the team's own horn (especially @alejoe91 and @magland as far as know), but if you are mostly doing merging `sortingview` compatibility is built into spikeinterface. It lets you curate with CCGs, waveform shape, and amplitudes and it customizable (last time I used it it didn't have PCA feature space so correct me if that has been added). But it has the added benefit of keeping track of the merges which are saved into a json (amongst other formats) which are then loaded into the spikeinterface objects. It's light and in the browser (and is super snappy compared to Phy). I'll leave the docs [here](https://spikeinterface.readthedocs.io/en/latest/modules/curation.html). Give them a read consider using it if it fits with your lab's general curation workflow.

closed 2024-02-08 by Antho2422