SpikeInterface/spikeinterface #2478: question on preprocessing followed by KS

paolahydra, 2024-02-12:
Hi everyone,

I am establishing a Spikeinterface pipeline for preprocessing (filtering, destriping) of NPX2 data.
The tutorials on these are very clear, so there are no issues so far.

To feed the preprocessed data to KS, I am currently saving the data with compression, as suggested by many tutorials.
However, there are two problems with this:
1. The preprocessed data is float32 and needs to be cast back to int16, but I could not find an example of this anywhere in the tutorials. What is the most efficient way to do it?
2. most importantly, there is a useless duplication of data since calling KS re-saves everything again **twice**: recording.dat and temp_wh.dat.

Perhaps the best approach is to compute motion correction and whitening, and directly save the temp_wh.dat to skip the preprocessing steps in KS as in [1418](https://github.com/SpikeInterface/spikeinterface/pull/1418/).

Is that correct? I would really appreciate some input on these steps, if not an actual tutorial. :)
Thanks a lot.
Paola

paolahydra, 2024-02-12:
Thanks, Alessio. 
Your answer was very useful. It also cleared up some confusion I had about the need to save the preprocessed data.

Perhaps I would add an option to delete the `recording.dat` copy after sorting is complete in `si.run_sorter_by_property` to mitigate the issue of limited disk space.

paolahydra, 2024-02-12:
super sweet! Thank you! :)

closed 2024-02-13 by alejoe91

JoeZiminski, 2024-03-11:
Hey @alejoe91 I was very interested to read your comments above, that's great to know. ATM I am typically filtering with a wider band in KS that used in SI preprocessing, and turn off CMR in KS, to minimize it. Do you think turning off CMR in kilosort is okay?

However I'd like to also start handling motion correction in SI, and eventually use whitening in SI too. Unless there is a workaround I don't know about, I find the KS output prohibitive as it's not always possible to reconstruct the data due to lack of saved whitening matrix, and so impossible to check preprocessing performance. However, have you found that replicating preprocessing steps in spikeinterface and switching all off in kilosort results in poor performance in sorting?

alejoe91, 2024-03-11:
@samuelgarcia can comment too, but we found that switching off ks preprocessing results in almost the same output if you do a similar preprocessing in SI (local whitening - which is default).

Pinging @oliche too, because we discussed about improving the whitening by making it better conditioned (which can be problematic for some brain areas , like cerebellum)