SpikeInterface/spikeinterface #2483: ipywidgets and Phy ProbeView channel locations mismatch

CGonzC, 2024-02-13:
Hi everybody,

I am confused since I noticed a mismatch in the channel maps generated in my Jupyter notebook and Phy reading the same kilosort2.5 sorting_output. For example, in the attached screenshots, two selected clusters are detected in channel 41 and 43 (42 and 43 in my OE). In the Jupyter notebook they are shown close together but in Phy far away. In the Jupyter widget the positions seem to coincide with the "contact id" of the probe, whereas in Phy they are shown in the expected position according to the OE channels.
I'm using a Cambridge Neurotech ASSY-77-H10, wired via ASSY-77>Adpt.A64-Om32_2x-sm-NN>RHD2164.

[image]
[image]

What do you think could be the problem here?
What am I missing or ignoring?
I would appreciate your help!
Thank you very much in advance
Cristian

alejoe91, 2024-02-13:
Hi,

The SpikeInterface unit IDs and ok phy IDs are not the same. See this issue: https://github.com/SpikeInterface/spikeinterface/issues/2426

To get the original phy IDs, you can do : `sorting.get_property("original_cluster_id")`

alejoe91, 2024-02-13:
Aldo, in the phy unit_table you should have an `si_unit_id` column if you exported with SI, with the SpikeInterface unit Id corresponding to the unit

CGonzC, 2024-02-13:
Hi Alessio,

Thank you very much! that was very fast!
I'm clear now :)
Thank you very much again!

Cristian

alejoe91, 2024-02-13:
No worries! Closing then :)

closed 2024-02-13 by alejoe91