SpikeInterface/spikeinterface #2508: `SortingAnalyzer.compute(x)` input types why not a list?

zm711, 2024-02-21:
A user can do 

`SortingAnalyzer.compute('waveforms')` as a string with no default supplied, but if I want to chain then I have to do
`SortingAnalyzer.compute({'waveforms': {}, 'spike_locations': {})` with a nested dict of empty dicts to just use the default.

Is there a reason why you don't just accept:
`SortingAnalyzer.compute(['waveforms', 'spike_locations'])`

It feels cleaner in the case of users that are just going to use the defaults anyway. I like the `dict` format to allow for super fine control, but is there a reason you don't want to give the non-power users a simple option for chaining?

samuelgarcia, 2024-02-26:
The reason is that we need dict to propagate kwargs to every extension. with the list this is not possible.

```python
SortingAnalyzer.compute('waveforms', **kwargs_wf)
SortingAnalyzer.compute('spike_locations', **kwargs_sl)

# is equivalent to
SortingAnalyzer.compute({
    'waveforms': kwargs_wf,
    'spike_locations' :kwargs_sl
})

```

samuelgarcia, 2024-02-26:
And also 
`compute()` is wrapper on top of `compute_one_extension()` `compute_several_extensions()`

zm711, 2024-02-26:
My point was more for new users/users who don't care about tuning the parameters (those that just trust you as the developer to have picked okay defaults). In that case I can do

```python
SortingAnalyzer.compute('waveforms') # uses defaults implicitly
```

But I can't do the same with multiple metrics. ie you require explicit defaults 

```python
{'waveforms': kwargs_wf,
 'spike_locations': kwargs_sl
}
```

I think explicit is better than implicit, but for the `compute_one_extension` you automatically fill in the defaults if not given but you don't for multiple. It could be as simple as if it is a list you just fill in the default for the user before putting it into `compute_several_extensions`. 

This all being said this is more a design choice rather than a real issue (implicit vs explicit in different functions). So we can close this, but I was just wanting to point out that there are two different ways right now that might confuse people who don't want the control.

samuelgarcia, 2024-02-26:
Oh yes . Sorry I read it too quickly. Yes we could accept list and convert it to dict with default params.
I will do this in the same PR

samuelgarcia, 2024-02-26:
Done in #2514

closed 2024-02-26 by zm711