SpikeInterface/spikeinterface #2515: Times not propagated by `.save_to_memory()`

cwindolf, 2024-02-26:
I was a bit surprised to see that the following is possible:

```
rec = si.read_binary_folder(...)
rec1 = rec.save_to_memory()
print(rec.sample_index_to_time(0), rec1.sample_index_to_time(0))
# => 2451.763033333333, 0.0
```

I'm working around this by `rec1.set_times(rec.get_times())`, but it seems to me that since `save()` maintains the times when saving to other formats (like the binary folder), it should do that for memory mode as well.

samuelgarcia, 2024-02-26:
Is it based on time vector or t_start ?

cwindolf, 2024-02-27:
This particular recording has t_start and no time_vector

cwindolf, 2024-02-29:
To get it working, I ended up doing
```
rec1._recording_segments[0].t_start = rec0._recording_segments[0].t_start
rec1._kwargs["t_starts"] = [rec0._recording_segments[0].t_start]
```

h-mayorquin, 2024-06-14:
Related:
https://github.com/SpikeInterface/spikeinterface/issues/2627

PR #3120 (2024-07-09): Fix `t_starts` not propagated to `save_to_memory`.
This PR closes #2515 by propagated `t_starts` to memory recording objects during `save_to_memory`. This is tested in #3117. 

This works to propagate the times to save to memory, and tests are passing, but I havn't considered other uses for these recording classes. So it would be great if others could let me know if this might break anything elsewhere! I can't see why it would as it uses default kwargs that maintain the old behaviour, but still I'm paranoid 😅 

Tests are added here that cover propagation of time information when stored as `t_s [... cut]
closed 2024-07-09 by alejoe91