SpikeInterface/spikeinterface #2516: Requesting - to_neo_spiketrain_list()

taningh86, 2024-02-26:
Hi,
It will be great if a function can be incorporated in spikeinterface modules to convert kilosort data into neo_spiketrain list. Currently, I believe in spiketrain only a function that takes neo_spiketrain list is available. This was discussed in Neo githib page as well https://github.com/NeuralEnsemble/python-neo/issues/1412

zm711, 2024-02-28:
Thanks @taningh86. The codebase is under a bit of a refactor at the moment, but this issue will serve as a reminder for @samuelgarcia to do when there is free time to work on it, unless you want to try to implement it yourself :)

zm711, 2025-04-04:
I'll try to put this on my todo list. Someone in my lab group was hoping for this now. :)