SpikeInterface/spikeinterface #2523: Quality metrics ISI

Antho2422, 2024-02-28:
Hi Spikeinterface team !

I have a question about ISI/RP violations metrics detailled here :   https://spikeinterface.readthedocs.io/en/latest/modules/qualitymetrics/isi_violations.html . 

I noticed that you implemented 2 different ways of getting isi violation ratios that seems complex compared to the naive way of getting the number of spikes within the refractory period : 
`ratio = np.sum(np.diff(spike_train) < t_r) / len(spike_train)`

I guess that the methods you implemented take into account more parameters like the refractory period enforced by the recording system and that makes sense. But I still don't get completly **the pros and cons about each metrics**. I also don't get exactly the way each metric should be interpreted to perform a curation. Because I was told and I read in multiple article that we usually consider a cluster being multi unit if the **ISI ratio was above around 1%**, but **I am not sure this threshold is valid for Hill and Llobet method**. 

Here is the comparison of the 3 different metrics (the "classic" one being the simple one detailled above) and you can notice that I get very different results : 

[image]

Can somebody knows a bit more about it and can help me figure out what are the pros and cons about using each method ? 

Thank you, 
Anthony

zm711, 2024-02-28:
@DradeAW, you have a strong backward in ISI right (and implemented (at least) one of the functions right?)

DradeAW, 2024-02-28:
@zm711 Indeed, I worked on the question and implemented the `compute_refrac_period_violations` function (shown here as the Llobet ratio)

You can find more detail in the pre-print we wrote (https://www.biorxiv.org/content/10.1101/2022.02.08.479192v1), but basically our formula (that's shown here as the Llobet ratio) makes the assumption that the spike train is made from a neuron (which can never violate its own refractory period) and some noise (contaminant spikes that happen at random). Following this assumption, we can derive mathematical equations and solve for the ratio of contaminant spikes.

The Hill ratio is similar, making the assumption that the noise comes solely from another neuron (so both neurons can violate each other refractory period, but neither neuron can violate its own). For low contamination, the two formulas converge. But they start being different after 15%

The one you detailed DOES NOT give the ratio of contaminant spikes, but only the ratio of refractory period violations (the contaminant spikes CAN occur in the refractory period of a spike, but do not HAVE TO).

Let me know if it's clear :)

Antho2422, 2024-02-28:
@DradeAW thank you for the clarification. It is a bit clearer but I'm still struggling to understand how it is possible to make this assumption without knowing from the spike train which spike is a "real" one and which one is a contamination. In other words, how can we identify the contaminant spikes occuring outside of the refractory period of the "real" spikes ?

Antho2422, 2024-02-29:
Okay so this is more an estimation on a theoritical/statistical point of view, I think I get it ! Thank you for the clarification !

rat-h, 2024-02-29:
@DradeAW, but you didn't answer a critical question @Antho2422 asked: Say Llobet ratio is 0.01. What can we say about the quality of the unit? Can we assess units based on Llobet or/and Hill ratio? I guess there will be some probabilities, so we can say that with 95% confidence, this unit is .... Without such classification, these ratios aren't really useful.

DradeAW, 2024-02-29:
The Llobet/Hill ratio gives you an estimate of the ratio of false positives in your spike train.

You're the one who decides what to do with it!
For example, from small neurons such as MLIs in the cerebellar cortex, I can never find one with less than 15% contamination ...
However for bigger neurons such as Purkinje cells, it can be quite easy to get some with <0.1% contamination

Depending on what you're doing, you decide what threshold you think is good to apply. What happens to your analysis if 10% of the given spikes are wrong? (keep in mind this is only the false positives. False negatives (i.e. missed spikes) don't show up!)

DradeAW, 2024-02-29:
No the question isn't stupid, it's normal to try to understand and question things. That's why we're here :)
However, I don't think it makes sense to have a null hypothesis "the unit is not contaminated".
The spike-sorting algorithms are never perfect, almost every single unit is contaminated. The question is how much? (I think we're fine with 0.01% contamination)

What could be possible is a confidence interval as you suggest, that we're 95% sure that the contamination is between C_min and C_max %.
But I don't really like that idea, because it is under the hypothesis that the contaminant spike source is random, which in some case is not true (e.g. complex spikes in the cerebellar cortex because of the gap junctions in the olivary neurons).

So personally, I like the metric as an estimation, and then a human behind knows if this is good for him or not (depends on the cellular type).

zm711, 2024-03-14:
I'll close this for now, but if there is more confusion around ISI/RP metrics reopen or open a new issue.

closed 2024-03-14 by zm711