SpikeInterface/spikeinterface #2525: Opening the raw file in matlab

JuanPimientoCaicedo, 2024-02-28:
Hi, I was wondering if you have an easy way to open a raw file in matlab. 

I have some code written in MATLAB that I would like to use in the raw data file. Of course I am not thinking about uploading the whole file at once, I was planning to use the memmapfile function to do it or something like that. 

[image]

thank you in advance.

JuanPimientoCaicedo, 2024-02-29:
Thank you for your response, I used Spikeinterface to demux the data and to perform drift correction. I have developed a pipeline in Matlab and right now I just need to upload the binary file. 

The initial number of channels of the dat file exported by SpikeGLX is 384, however, after eliminating bad channels and performing the drift correction I end up with less channels, generally 5 to 6 less channels.

I am not sure if the channels are really eliminated from the file or are just ignored by Spikeinterface when doing subsequent analyses. 

Here you can see how I generally create the memory map in matlab.
```
filename = 'path\to\the\file'
dataType = 'int16'
channels = % Number of channels in the recording
nSamples = % Number of samples per channel
mmf = memmapfile(fileName, 'Format', {dataType, [channels nSamples], 'x'});
```

I am indeed planning to start doing more things in python, you have a pretty elegant library.

zm711, 2024-02-29:
To get most of that information you can open the `binary.json` in the folder. It will say which channels are include as well as number of channels. It will also say the dtype. Then you could calculate samples based on the file size and number of channels. It will also include the gain and offset. These are important if you want to convert your data to real voltages for your later analysis. Does that make sense?

zm711, 2024-02-29:
One last important point is 'c' order vs 'f' order. I'll have to tag in @alejoe91 to weigh in on the data. I would guess "c" ordered, but I don't want to say the wrong thing. I would imagine that we would do [samples channels] instead of [channels samples]. But let's wait for @alejoe91 to weigh in .

alejoe91, 2024-02-29:
Correct! The shape is [samples, channels]

JuanPimientoCaicedo, 2024-02-29:
Thank you for your response, it helped a lot!

closed 2024-02-29 by JuanPimientoCaicedo