SpikeInterface/spikeinterface #2531: extracting waveform from the Broadband

JuanPimientoCaicedo, 2024-03-01:
Hi, I am currently analyzing waveforms from spikesorted data with Neuropixels. I will briefly explain the pipeline:

1. I record in broadband mode in spikeGLX.
2. Preprocess the data with Spikeinterface.
3. Spike sort with Kilosort 3.0.
4. Manually curate in phy.
5. Upload the phy sorting object to spikeinterface again and use the function extractwaveforms.

It is in this final step where I am getting problems, given that I am working in extracting waveforms from both the highpass filtered data and the broadband data.

When I obtain the templates from the highpass flitered data everything works as expected, I get good waveform averages and metrics. However, when I try to do it with the broadband, I am finding that some units are correctly extracted while others don't.

Here you have an example of a successful extraction in both bands.

[image]

[image]

However, some units do not appear when I try to extract them in the broadband data.

[image]

[image]

Here I am attaching the code I am developing, I know it is not very well organized. So I am willing to explain any step if necessary. thank you!

[Postprocessing code.zip](https://github.com/SpikeInterface/spikeinterface/files/14464081/Postprocessing.code.zip)

samuelgarcia, 2024-03-04:
Hi.
The main reason is that if you give the non filtered recording to waveform extarctor then it try internally to estimate the sparstity with few spikes (100 per units). On the filetred version the max channel is easy to discover and so the sparsity mask is OK. But on the full band you can get mistake in sparsity estimation.
One solution is to get the sparsity from the filetred waveform and give it to extarct waveforms for the full band.
You should have better results.

zm711, 2024-03-20:
@JuanPimientoCaicedo,

do you try Sam's suggestions and figured this out? Can we close this issue or did you have more questions regarding the broadband?

closed 2024-03-20 by JuanPimientoCaicedo