SpikeInterface/spikeinterface #2534: preprocessing and spykingcircus2

sofiapuvogelvittini, 2024-03-04:
Hi, thank you very much for this package!
I am working with Maxwell data and sorting with spikeinterface , sorter_name="spykingcircus2".
According to my understanding, spikyng circus should receive unfiltered data as input, becuase it will do the filtering steps internally. Therefore, I've skipped the filtering steps suggested in the SpikeInterface tutorial.   However, I am doubting now if this is the correct approach when using spykingcircus2 inside spikeinterface. 

I've encountered an issue when attempting to "extract waveforms" using si.extract_waveforms(). The exception message indicates that the recording is not filtered and advises using 'bandpass_filter()'. 
Do you have more specific guidelines or advice on this matter, especially when sorting with with spykingcircus2?

Thanks for your help!

zm711, 2024-03-04:
Howdy @sofiapuvogelvittini,

So when extracting waveforms you should extract on filtered data. So even if you don't filter the data for a particular sorter once you get the sorting data, you should do the extraction of the preprocessed recording instead (a typical workflow would be bandpass + CMR/CAR). If you really want the raw unfiltered data you would need to extract with `allow_unfiltered=True` instead.

Again a normal workflow would do extraction on preprocessed data, but it depends on your purposes if you want to not do this.

Does that make sense?

yger, 2024-03-05:
Hi. I'm on holidays so I ll be quick. Internally, SC2 does a highpass (150Hz), then a common reference, and a zscore if apply_preprocessing is set to True. But this does not modify the recording given to SC2 , so if you want to extract waveforms after, you need to redo a similar preprocessing. Bypassing the preprocessing if meant to be for users that would have special steps, such as moron correction for example.

sofiapuvogelvittini, 2024-03-06:
Thanks yger for this explanation! enjoy the holidays

zm711, 2024-03-06:
Great we will close this then. But feel free to open new issues if new questions come up :)

closed 2024-03-06 by zm711