SpikeInterface/spikeinterface #2539: se.read_maxwell with MaxTwo data

sofiapuvogelvittini, 2024-03-04:
Hello, I'm currently working with spikeinterface using Maxwell data. While I haven't encountered any issues reading MaxOnw data, I'm facing a problem with MaxTwo data.

My objective is to read the well (or stream_id) well006 from my .h5 file. Upon inspecting the h5 file in Matlab, I observed only one recording associated with that well, and the pointerToData indicates:
'/wells/well006/rec0000/groups/routed/raw/'   (attached a screenshot) 
[pointerRoData.pdf](https://github.com/SpikeInterface/spikeinterface/files/14485215/pointerRoData.pdf)

As a result, I attempted the following command:
full_raw_rec= se.read_maxwell('/mnt/data/spike_sorting/nikki/prep_data/nikki_data.raw.h5',
                              stream_id='well006', rec_name='rec0000')

However, I encountered the error: KeyError: "Unable to synchronously open object (object 'rec0001' doesn't exist)"

I am not sure why the program is attempting to find rec0001 when I've specified the parameter rec_name='rec0000'

Could you please provide some guidance on resolving this issue? Thanks a lot.

Full error: 

full_raw_rec= se.read_maxwell('/mnt/data/spike_sorting/nikki/prep_data/nikki_data.raw.h5',
                              stream_id='well006', rec_name='rec0000')

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[33], line 1
----> 1 full_raw_rec= se.read_maxwell('/mnt/data/spike_sorting/nikki/prep_data/nikki_data.raw.h5',
      2                               stream_id='well006', rec_name='rec0000')

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/neoextractors/maxwell.py:58, in MaxwellRecordingExtractor.__init__(self, file_path, stream_id, stream_name, block_index, all_annotations, rec_name, install_maxwell_plugin)
     55     self.install_maxwell_plugin()
     57 neo_kwargs = self.map_to_neo_kwargs(file_path, rec_name)
---> 58 NeoBaseRecordingExtractor.__init__(
     59     self,
     60     stream_id=stream_id,
     61     stream_name=stream_name,
     62     block_index=block_index,
     63     all_annotations=all_annotations,
     64     **neo_kwargs,
     65 )
     67 self.extra_requirements.append("h5py")
     69 # well_name is stream_id

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:187, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    158 def __init__(
    159     self,
    160     stream_id: Optional[str] = None,
   (...)
    165     **neo_kwargs: Dict[str, Any],
    166 ) -> None:
    167     """
    168     Initialize a NeoBaseRecordingExtractor instance.
    169 
   (...)
    184 
    185     """
--> 187     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    189     kwargs = dict(all_annotations=all_annotations)
    190     if block_index is not None:

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:27, in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     23 def __init__(self, block_index, **neo_kwargs):
     24 
     25     # Avoids double initiation of the neo reader if it was already done in the __init__ of the child class
     26     if not hasattr(self, "neo_reader"):
---> 27         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     29     if self.neo_reader.block_count() > 1 and block_index is None:
     30         raise Exception(
     31             "This dataset is multi-block. Spikeinterface can load one block at a time. "
     32             "Use 'block_index' to select the block to be loaded."
     33         )

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:66, in _NeoBaseExtractor.get_neo_io_reader(cls, raw_class, **neo_kwargs)
     64 neoIOclass = getattr(rawio_module, raw_class)
     65 neo_reader = neoIOclass(**neo_kwargs)
---> 66 neo_reader.parse_header()
     68 return neo_reader

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/neo/rawio/baserawio.py:185, in BaseRawIO.parse_header(self)
    172 def parse_header(self):
    173     """
    174     This must parse the file header to get all stuff for fast use later on.
    175 
   (...)
    183 
    184     """
--> 185     self._parse_header()
    186     self._check_stream_signal_channel_characteristics()
    187     self.is_header_parsed = True

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/neo/rawio/maxwellrawio.py:109, in MaxwellRawIO._parse_header(self)
    107     self._channel_slice = ids
    108 elif int(version) > 20160704:
--> 109     settings = h5["wells"][stream_id][self.rec_name]["settings"]
    110     sr = settings["sampling"][0]
    111     if "lsb" in settings:

File h5py/_objects.pyx:54, in h5py._objects.with_phil.wrapper()

File h5py/_objects.pyx:55, in h5py._objects.with_phil.wrapper()

File ~/anaconda3/envs/si_env/lib/python3.11/site-packages/h5py/_hl/group.py:357, in Group.__getitem__(self, name)
    355         raise ValueError("Invalid HDF5 object reference")
    356 elif isinstance(name, (bytes, str)):
--> 357     oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
    358 else:
    359     raise TypeError("Accessing a group is done with bytes or str, "
    360                     "not {}".format(type(name)))

File h5py/_objects.pyx:54, in h5py._objects.with_phil.wrapper()

File h5py/_objects.pyx:55, in h5py._objects.with_phil.wrapper()

File h5py/h5o.pyx:189, in h5py.h5o.open()

KeyError: "Unable to synchronously open object (object 'rec0001' doesn't exist)"

zm711, 2024-03-05:
@sofiapuvogelvittini,

Spikeinterface uses neo under the hood for a lot of the ephys reading. Could you open the issue there since it would need to be fixed there? 
https://github.com/NeuralEnsemble/python-neo/issues

zm711, 2024-05-10:
We fixed this in Neo: https://github.com/NeuralEnsemble/python-neo/pull/1444

Which is now included in a Neo release. So closing this. Feel free to open new issues if they come up :)

closed 2024-05-10 by zm711