SpikeInterface/spikeinterface #2540: Too Slow processing- sw.plot_unit_templates(we, sparsity=sparsity, backend ="ipywidgets")

taningh86, 2024-03-04:
Hi,
I am analyzing NPX 2.0 data and finding that the resulting waveform plot from the code below is very slow to toggle between units. It was not the case with NPX 1.0 data and toggling between units resulted in instant display of respective waveforms.
`sw.plot_unit_templates(we, sparsity=sparsity, backend ="ipywidgets")`

I don't think what probe the data comes from should make a difference? 
I upgraded to the latest version (0.100.0) and wonder if that could be the cause. The reason for upgrading was 0.99.1 version I had before was not able to analyze NPX 2.0 data. 
Please let me know if there's a trick to making the processing faster. 
For comparison, sw.plot_amplitudes(we, backend="ipywidgets") shows no issues and processing is instantaneous.

Thanks
Jimmy

taningh86, 2024-03-05:
Hi Zach!
Yes, its pretty slow. I get the logic of waveforms being more processing intensive than amplitudes, but I did not observe them this slow with NPX 1.0 data and the last version 0.99.1 version. When I say slow its taking close to 30 sec-1 min for a unit. With NPX 1.0 data it was immediate, just like the amplitudes plot. Having said that, I have not worked with NPX 1.0 data with version 0.100.0. 
I am using num_channels =4 using the method = "best_channels". I am unable to use method = "radius" , with um > 10. I am getting "assertionError" with that. I will raise a different issue for this. 
I am plotting the waveforms with the template I think. May be that could be causing the delay. Below is the plot output:
[image]

alejoe91, 2024-03-05:
@taningh86 I think that the extra time is because of the shading. By default, for each unit we extract the 1st, 25th, 75th, and 99th percentiles to plot two different shadings. Can you try to set `template_percentile_shading` to `None` and see if this recovers the previous performance? Note that in this case a shading of +- std will still be plotted, but that is precomputed so it shouldn't add the extra computation!

taningh86, 2024-03-05:
Hi Alessio,
I just tried that 
`w = sw.plot_unit_templates(we,template_percentile_shading = None, backend="ipywidgets")`

But throws a 'not a valid backend keyword argument'.

taningh86, 2024-03-05:
May be you meant set 'template_percentile_shading = None' in the code below?
```
we = si.extract_waveforms(recording_saved, sorting, folder=base_folder / "waveforms", allow_unfiltered=True,
                          load_if_exists=False, overwrite=True, **job_kwargs)
```

alejoe91, 2024-03-05:
No, in `plot_unit_templates` use `templates_percentile_shading=None`

There was a typo on my side!

taningh86, 2024-03-05:
Got it! And that did it!! Thanks Alessio. We can close this issue now.