SpikeInterface/spikeinterface #2544: How to get started

mniazy1, 2024-03-05:
I am a beginner in using this, and I have no prior knowledge in this field. Currently, I just installed Spike Interface into Anaconda, and I want to move into testing and getting familiar with the interface. However, I am not sure how to get started with "How to 'get started'." Furthermore, I'm not sure if everything is installed fully since I followed the beginners Installation tips, (the cleanup_for_windows.py is not working / invalid). 

I have the Anaconda command prompt open into (si_env) cd'd into Documents. What I want to know is where do I run the commands / code from the How to get started tutorials, such as an Import for example. Do I use Spyder from Anaconda? Or through the Anaconda command prompt (si_env)? Or is it somewhere else to run and operate everything.

zm711, 2024-03-06:
Howdy @mniazy1,

What do you mean it is not working? What specific error is it giving? We can better diagnose what's going of if we know exactly what is happening. The `check_your_install` can just be run in the command prompt. Ie you should do

```bash
conda activate si_env
python check_your_install.py
```

Then it should run through some testing. If this fails let us know. When it finishes you should do:

```bash
python cleanup_for_windows.py
```

If you want to then run code with the installation_tips env, we have it install Jupyter and so you can create a Jupyter notebook and run any of the tutorials from there. If you want to use Spyder then you would either need to include it in the env (more beginner friendly less storage friendly) or work with ipython kernels (more storage friendly less beginner friendly). 

Just so we can help you the best possible are you a python beginner or a spike sorting beginner? Or just a conda beginner?

mniazy1, 2024-03-18:
Hello @zm711, sorry for the quite late reply!

For the "cleanup_for_windows.py" in the installation process (step 5), the file itself shows this when clicked on "400: Invalid request." And it is not downloadable as per the steps, so running the second command you showed does not work.

So you're saying after installing Spike Interface and activating the environment, I can then perform the "how to get started" steps through Jupyter notebook from Anaconda?

I have a bit of python experience but I would not say I am an intermediate. As for spike sorting I have 0 previous knowledge and have not worked with it at all.

Thank you!

zm711, 2024-03-18:
No worries @mniazy1 :) 

I just fixed the bad link. Thanks for finding that. If you want you can clone the whole repo with git or you can just open that file and copy the contents into a `.py` file yourself and then run it locally or you can wait until we merge that fix and then you can download the fixed version.

Yes in that environment (if you install with [`full_spikeinterface_environment_windows.yml`](https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/full_spikeinterface_environment_windows.yml)). You can use a juypter notebook if you want. You'll just need to make sure that you have the correct environment kernel in your notebook (a quick google found [this](https://saturncloud.io/blog/how-to-set-a-default-environment-for-anaconda-and-jupyter/#:~:text=Setting%20the%20Default%20Environment%20for%20Jupyter&text=Select%20the%20default%20environment%20kernel,from%20the%20kernel%20selection%20menu.) to help with the process, but you could google yourself if you want).

We have a lot of docs and tutorials that can help with the beginner pythonista so definitely read our docs and open issues as you run into problems. As far as no previous spike-sorting our docs also cover some of that so I would take a few days and really just read our docs in detail. If you run into questions you can open an issue if you need to. 

How does that sound?

mniazy1, 2024-03-18:
Hello @zm711,

I was able to set everything up properly, now I just have to start testing. But before I do I plan on reading the docs but I'm not entirely sure what it is you're referring to. Is it the 200 page PDF file found in the main Spike interface website, under the "Read the docs: Downloads"? Or is there more?

zm711, 2024-03-18:
Great to hear that you've got things going. No I don't think we have updated the pdf. The pdf is for version 0.13 and the current version is 0.101.0 with the sorting analyzer or 0.100.0 with the legacy waveform extractor. It might actually be helpful to know which version you downloaded. Could you go to your terminal and type:

```
(si_env) $ python
>>> import spikeinterface
>>> print(spikeinteface.__version__)
```

Just so I can point you to the correct docs. For version 101 go [here](https://spikeinterface.readthedocs.io/en/latest/overview.html). For version 100 go [here](https://spikeinterface.readthedocs.io/en/0.100.1/overview.html). 

Either way we currently just have the docs on line without the pdf option, so just tab through the docs through. There are some tutorials you can run, there is a modules gallery that shows tiny examples, and then the module documentation that goes in depth for each part of the code base. So check those out!

mniazy1, 2024-03-18:
I have version 0.99.1, should I read [this](https://spikeinterface.readthedocs.io/en/0.99.1/overview.html#), or do you suggest updating it, if so how can I go on about it.

zm711, 2024-03-18:
@mniazy1,

honestly updating is probably for the best. There are still a bunch of bug fixes and speed improvements happening all the time. You could do

```
(base) $ conda activate si_env
(si_env) $ pip install -U spikeinterface
```

That should bring you up to 0.100.1. Even better would be to install from source and just learn 0.101.0. The main difference between the two is that 101 has the new SortingAnalyzer object that allows for faster calculations, better IO (WIP) for things that require the Recording object, and an easier API. That being said we have a translation class so even if you upgrade to 0.100.1 and start analyzing data you can still upgrade later.

Let me know if you want help installing from source otherwise just type the command I put above to get the most recent release.

mniazy1, 2024-03-18:
I think I'll stick with 0.100.1 for now since I am just learning about how use the interface. Also, just to make sure I'm on the right track, I am currently following the "[How to get started](https://spikeinterface.readthedocs.io/en/0.100.1/how_to/get_started.html)" guide, and is this what my Jupyter notebook should like like while implementing the steps and lines of codes? Will I be able to run Spike Interface like this and perform analysis?

[image]

zm711, 2024-03-18:
It looks good to me. I would give it a go and as long as there are no import errors you should be ready to try. 

Note that some of our tutorials require Datalad to get running which can be a bit of a pain. So ideally rather than download sample data I would recommend trying to put your own data in and start analyzing your own stuff :)

mniazy1, 2024-03-18:
Is it possible to use my own data in the tutorials / how to get started guide instead of the sample data? Because I need to first learn how to even use Spike Interface and how it works before I can start analyzing my own data efficiently and in details.

zm711, 2024-03-18:
It's pretty easy to start thinking about this. Let's say your data comes from Intan. In a tutorial when it says recording you would just do:

```python
recording = se.read_intan('path-to-data/name-of-file.rhd', stream_id='0')
```

Then you use your data for the tutorial. Granted you'd also need to make a probe. But often times if you're actually trying a tutorial it helps to also try your data. When you get to the probe section you try to make a probe and see if you can figure it out and if you can't you go to the ProbeInterface docs and learn there too :)

Since you're new to spike-sorting realizing you even need a probe might be new. So you should also look at our [probe docs](https://probeinterface.readthedocs.io/en/main/) too to understand how probes and their wiring work.

If you don't feel ready yet to try to use your own data you can definitely try to use some of the sample data stuff, but that is easy for some tutorials and harder for others so up to you. :)

zm711, 2024-03-20:
@mniazy1,

I'll close this issue as completed since we've got you started. If you run into new problems with the API etc feel free to open new issues on the specific problem

:)

closed 2024-03-20 by zm711