SpikeInterface/spikeinterface #2555: frame slice lost spike train on merged units

yuhang21, 2024-03-07:
Hi I encountered an issue that I lost the spike train for the merged units when I extract the waveform from the sliced sorting from a concatenated sorting

I used the following line the generate a concatenated _rec for sorting using mountansort5.
`concatenated_rec = si.concatenate_recordings(recordings_list)`
I used the sortingview to postprocess and merge my units
`si.plot_sorting_summary(we_curated_auto, curation=True, backend='sortingview')`
The final sorting result from the sorting result looks fine, with proper amount of spike train in the merged units.
However, when I sliced the rec and sorting using the `.frame_slice` and extract the waveform again using `si.extract_waveforms` then all the spike trains for the merged units are gone. The 'sample_index_94.npy' and 'waveroms_94.npy' (as an example) are only 1kb with no spike trains in them. I copy the output. 

> spike_train = segment_sorting_1.get_unit_spike_train(unit_id=94) 
print(spike_train)
[]
spike_train = sorting_curated_final.get_unit_spike_train(unit_id=94)
print(spike_train)
[     5162      9173     10890 ... 598744711 598787285 598789000]

I tried to copy the metadata but didn't help `sorting_curated_final.copy_metadata(segment_sorting_1)`

I am wondering whether I missed something. I tried to utilize the `spikeinterface.core.split_sorting` but no list as an output I can call for a separate waveform extractions. Thank you very much.

zm711, 2024-03-08:
Howdy @yuhang21, could you post your whole script. It is hard to figure out if there is a problem without knowing everything you did.

yuhang21, 2024-03-09:
Hi thanks for getting back. Let me copy my pipeline here

load the data and concatenate_rec

```
binary_file_path_1 = 'E:/EBL data/tracking_me/day_1.dat'
binary_file_path_2 = 'E:/EBL data/tracking_me/day_2.dat'
rec1 = se.BinaryRecordingExtractor(binary_file_path_1, sampling_frequency=30000, num_channels=128, dtype='int16')
rec2 = se.BinaryRecordingExtractor(binary_file_path_2, sampling_frequency=30000, num_channels=128, dtype='int16')
recordings_list = [rec1, rec2]
concatenated_rec = si.concatenate_recordings(recordings_list)
rec= concatenated_rec
```
preprocessing and sorting 
```
rec_filt = sp.bandpass_filter(
    rec, freq_min=300, freq_max=6000, dtype='int32')
rec_cr = sp.common_reference(
    rec_filt, operator="median", reference="global")
rec_preprocessed = sp.whiten(rec_cr, dtype='float32')
rec_for_wvf_extraction = rec_cr 

sorting = si.run_sorter(
    sorter_name='mountainsort5',
    recording=rec_preprocessed,
    output_folder=Path(save_folder) / 'sorting',
    remove_existing_folder=True, 
    verbose=True,
    **ms5_params
)
```

Then extract waveforms, perform merge, and get final result 
```
we = si.extract_waveforms(rec_for_wvf_extraction, sorting, folder=save_folder, 
                          mode="memory", ms_before=1, ms_after=2, overwrite=True, 
                          sparse=False, max_spikes_per_unit=50000,**job_kwargs)
si.plot_sorting_summary(we, curation=True, backend='sortingview')
manual_curation_sorting = scur.apply_sortingview_curation(
    sorting_curated_auto,
    uri_or_json=uri,
    skip_merge=False)
keep_idx = np.where(manual_curation_sorting.get_property("accept"))[0]
discard_idx = np.where(manual_curation_sorting.get_property("reject"))[0]
remove_unit_ids2 = unit_ids[discard_idx]
sorting_curated_final = manual_curation_sorting.remove_units(remove_unit_ids2)

```

Then I start to slice the frame
```
sorting_segments=si.split_sorting(sorting_curated_final,recording_or_recording_list=concatenated_rec)
waveform_extractors = []
for i in range(2):
    start_frame = sorting_segments._sorting_segments[i].start_frame
    end_frame = sorting_segments._sorting_segments[i].end_frame
    segment_sorting = sorting_curated_final.frame_slice(start_frame=start_frame, end_frame=end_frame)
    rec = rec_for_wvf_extraction.frame_slice(start_frame=start_frame, end_frame=end_frame)
    sorting_curated_final.copy_metadata(segment_sorting)
    segment_sorting.save(folder=Path(save_folder) / f'segment_sorting_{i}', overwrite=True)
    
    we_sorting = si.extract_waveforms(rec, segment_sorting, folder=Path(save_folder) / f'waveforms_sorting_{i}',
                                     overwrite=True, ms_before=1., ms_after=2., sparse=False, max_spikes_per_unit=None,
                                     **job_kwargs)
    waveform_extractors.append(we_sorting)
```
Then I found some of my merged units have gone, only left the name and index but no spike trains in.

zm711, 2024-03-11:
@alejoe91, you want to take a look at this. I don't use the slicing mechanism a lot so it might be better for you to verify it was done correctly.

alejoe91, 2024-03-11:
@yuhang21 

can you `print(start_frame, end_frame)` in the slicing loop?

BTW, you don't need the `copy_metadata()` call because the `frame_slice` does that by default. (see [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/core/frameslicesorting.py#L84))

Also, for the loop, it's better to use `for segment_index in range(sorting_segments.get_num_segments())`

zm711, 2025-02-21:
I will close this for being stale since we never heard. Feel free to reach out again with additional questions if we can be helpful!

closed 2025-02-21 by zm711