SpikeInterface/spikeinterface #2556: running kilosort2 in docker produces "EIG did not converge at index"

mgraupe, 2024-03-08:
Hello, 

I am using SpikeInterface version 0.100.0 and I am using the sorter kilosort2 in the docker on Ubuntu (22.04.2 LTS).  I launch the spike sorting with 

```
sorting = si.run_sorter(sorter_name='kilosort2', recording=recCG, output_folder= 'kilosort2_spkinterface/',
                             docker_image=True, verbose=True)
```

The sorting aborts with the error 
```
----------------------------------------Error using gpuArray/eig

EIG did not converge at index = 557499.
Error running kilosort2
```
This error can be resolved by downgrading matlab to 2021b or by replacing the line 100 in mainLoop/runTemplates.m to:

```
 try
        [A, B, C] = svdecon(UA);
 catch
        [A, B, C] = svd(UA,"econ");
  end

```
Is there a possibility to fix that that in the docker image? 

Thanks in advance. 

Best,
Michael
[image]

samuelgarcia, 2024-03-08:
Hi Michael.
thank you for the feedback.

Unfortunatly this is a long task.

First, we need to submit this patch to kilosort2 branch and make it accepted.
Then, on spikeinterface side,  we have to compile this kilosort2 code and embed it into a dock image and then push the image.

I propose you submit, you try to sumbit the patch on killosort side, when accepted we try to propagate it here.
I must admit that building docker images is really tedious and time consuming.
Alessio and You Min did the most of the work.

samuelgarcia, 2024-03-08:
and why not using kilosort2.5 ? The one wich best bechmarks results.

zm711, 2024-03-08:
Not to throw a wrinkle into this, but @DradeAW tried to submit a patch for KS and they closed it (because they seem to be done patching anything Matlab related). So I highly doubt patching at the KS level will happen. As far as 2 vs 2.5--> the docs for KS say low channel count may still be better with 2 vs 2.5. So for tetrodes or other small electrodes maybe 2 would still be better. Not sure though.

zm711, 2024-04-29:
@alejoe91 would there be any interest in building a docker with a different version of matlab or should we close this as not possible? I can confirm locally that Matlab 2021 fixes the issue and that later versions of Matlab will create this issue (not always always, but very often).

alejoe91, 2024-04-29:
Thanks for pinging @zm711 

This is very tricky on my side since I don't have access to other MATLAB versions :(

I suggest we close this

zm711, 2024-04-29:
I'll close this then. @mgraupe see thread for why this isn't currently possible. If you have access to a local copy of matlab you can either use 2021 or earlier or implement the fix that @DradeAW submitted [here](https://github.com/MouseLand/Kilosort/pull/560).

closed 2024-04-29 by zm711