SpikeInterface/spikeinterface #2557: SortingAnalyzer zarr format json serialization issue

zm711, 2024-03-08:
I tried to make a zarr SortingAnalyzer with a  `UnitsSelectionSorting` and a bandpassed and cmr'ed recording.

```python
si.create_sorting_analyzer(sorting_selected, rec_cmr, format='zarr', folder=folder)
```

Error occurs here in spikeinterface

```python

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:435 in create_zarr
    zarr_root.create_dataset("sorting_provenance", data=zarr_sort, object_codec=numcodecs.JSON())
```
with a final error of:
```
    raise TypeError(f'Object of type {o.__class__.__name__} '

TypeError: Object of type int64 is not JSON serializable
```

The rest of the stack trace is all within zarr.

If I do format='memory' it works fine. And format='binary_folder'. Also works fine. I'll go with binary for now, and will see if I can track down the problem (maybe this is a windows issue....).
@samuelgarcia (not sure if people are going to use Windows at your workshop so this might be important to work on asap if you plan on using zarr).

alejoe91, 2024-03-08:
@zm711 this should be easu to fix. Can you try the following:

```
from .core_tools import check_json

...

zarr_root.create_dataset("sorting_provenance", data=check_json(zarr_sort), object_codec=numcodecs.JSON()
```
?

samuelgarcia, 2024-03-08:
I the sorting is no json serilizable we should switch to pickle like do at many place.

zm711, 2024-03-08:
It works with check_json. There is still a lot of json serialization in SortingAnalyzer (at least that I can see). I like the json for going back and reading it, but if pickle is faster/easier than I'm down for whatever. I did find a couple other bugs that I put into that PR, so if you want to look at that Sam you could merge and then remove json in a bigger PR that you make whenever.

samuelgarcia, 2024-03-08:
Forget my comment the switch Json then pickle was already in the code.
The patch you did is ok for me

PR #2559 (2024-03-11): Add `check_json` to `zarr` for `SortingAnalyzer` for sorting_provenance file writing
Fixes #2557 

@alejoe91's fix for writing the sorting_provenance
closed 2024-03-11 by alejoe91