SpikeInterface/spikeinterface #2560: `SortingAnalyzer.Extension` Feedback

zm711, 2024-03-09:
So after using the extensions some more I have a couple ideas that I wanted to jot down just to think about.

One the error messaging isn't super helpful right now for two reasons. One I think we are all so use to the design of the package that when I try to run `compute('quality_metrics')` and it gives me an error I just import quality metrics because I know that's where it lives (ie I know the codebase). But I can imagine a new user doing `compute('spike_locations')` getting an error and not know which submodule to import. This would be super frustrating. So I'm wondering if we could add an attribute to extensions or maybe just a boolean when checking the extensions such that (in pseudocode)
```
if `quality_metrics` in ext_name:
    error_msg += 'import spikeinterface.qualitymetrics'
else:
    error_msg += 'import spikeinterface.postprocessing'
```

Problem 2 related to this issue is. If I type `.compute('quality_metrics')` without import qualitymetrics or I try `.compute('fake_extension')` I get the same error which is super confusing. I don't know what the solution is here since you're trying to allow people to make their own extensions. But it makes it hard to know what is actually possible or not with the package. (What *a priori* knowledge should people have of what is possible... I guess the docs could help with this a little bit). I'm really not sure what the solution is, but I discovered this when I mistyped `compute('quallity_metrics')` and it failed to register even though my typo clearly doesn't exist I was super confused until I read carefully.

Finally,
when reloading I was trying to be super light so I only imported core. But I had extensions: [spike_locations, spike_amplitudes, quality_metrics]. Now these didn't load (because I hadn't imported the correct submodules yet). At first I was confused because I knew I had just calculated these values and (had accidentally restarted my computer) so I knew they were there. I checked my extensions folders and they were there. Then it dawned on my that I need to import the modules to be allowed to load my old data. Once I'm use to the new workflow that'll be no big deal, but I'm worried that new users will have issues with "my data is there but I can't get it to load." So is there some way we could scan the extensions folder and warn them that not all data is loaded because not all submodules were imported? (In this case if the extension exists and was saved the user may want to reload it and not realize they need the submodule in order to be allowed to reload it). And If I then import the new submodules can I refresh my `sorting_analyzer` so that it loads the missing extensions? (That's an honest question--I could look in the source code myself, but since I'm already writing a novel I might as well finish). 

But seriously, the new sorting_analyzer has such a better flow and feels so much better to use. I'm really excited about this change! I think everything is smoother and with the pipeline improvements I can imagine this being quite a bit faster to when having to deal with the IO stuff.

samuelgarcia, 2024-03-11:
Salut Zach,
thank you for this is feedback.
We are aware of this and already discuss a bit with Alessio. Having better error message is of course necessary.
But this would need a central place in core for "known extensions" to display from which module they come from.
I am not sure to like this design but this will be necessary so lets maturate this a bit.
If we have this centralized list, we could also think of an auto load module when loding extension back so the user should not have to worry of importing the module. Also the extension could declare in the stored json from which module they come from. Lets see what it the easiest.

Note that for postprocessing, this is the main reason I advice to use `import spikeinterface.full as si` because the use do not have to worry about module stuff and extension will always work (compute and reload).

alejoe91, 2024-03-11:
I think this should work regardless of `spikeinterface.full`!

samuelgarcia, 2024-03-11:
But by design we have several sub module and if the sub module is not imported then the class cannot be computed/reload. We cannot fight against that.
The main question is more how we have to make a good error message or maybe make an automatic module import.
The automatic load would mean that the core is aware of all extensions that exists in others modules which is a design I do not like.

I was just claiming that the `spikeinterface.full` is the best solution for end user to not worry about the import. Even more for the postprocessing because the computation depend on it.

zm711, 2024-03-11:
For me I was thinking of a really naive approach of when you search through extension names you just add in a check if the string `quality_metrics` is listed in the compute function then tack on qualitymetrics and if not then tack on post processing to the error message.
Something like:

```python

    if extension_name == 'quality_metrics':
        error_msg = 'import spikeinterface.qualitymetrics'
    else:
        error_msg = 'import spikeinterface.postprocessing'
    global _possible_extensions
    extensions_dict = {ext.extension_name: ext for ext in _possible_extensions}
    assert (
        extension_name in extensions_dict
    ), f"Extension '{extension_name}' is not registered, please import related module before use" + error_msg
    ext_class = extensions_dict[extension_name]
    return ext_class
```

zm711, 2024-03-21:
I'm going to close this as the auto-import PR fixes the 3 issues listed here.

closed 2024-03-21 by zm711