SpikeInterface/spikeinterface #2565: no sortinganalyzer.py in the spikeitnerface=0.100.1 tag

rohanmalige, 2024-03-11:
Hi,

I and @mandarmp recently updated the spikeinterface module to 0.100.1 unfortunately when I run it I get the error "spikeinterface.full cannot find create_sorting_analyzer". I think the issue is because of the sortinganalyzer.py file not being present in the latest tag.

Can you please advise on the next steps?

Best,
Rohan

alejoe91, 2024-03-11:
Hi Rohan,

The 0.100.1 is a patch release for 0.100.0. The Sorting analyzer will be release in the next major release (0.101.0)

rohanmalige, 2024-03-11:
Thank you for the update, we got confused with the main branch in the spikeinterface-gui asking us to use the new create_sorting_analyzer. Thanks for the clarification, I will go ahead and close the issue.

closed 2024-03-11 by rohanmalige

samuelgarcia, 2024-03-12:
Oups. Sorry. Yes the spikeinterface-gui page switch to the SortingAnalyzer in the main branch.
You need to take tag 0.8.0 to go back to WaveformExtractor.