SpikeInterface/spikeinterface #2569: Cannot get KS4 to utilize my GPU

mohebial, 2024-03-12:
Hi,
    I am trying to get KS4 to work on spikeInterface to sort long NPX1.0 recording sessions. Whether I use the docker image or a local KS4 installation, my GPU is not engaged, and processing takes forever to complete. It shouldn't be my CUDA settings, since I can run standalone KS4 just fine using the GPU (RTX 3060). Do you have any suggestions on what might be wrong with my spikeInterface setup? I am running the following script in Jupyterlab on a Windows machine.

Thanks!
Ali
```

# Define recording object using the local file address
raw_rec = si.read_openephys(base_folder,stream_id='0')
rec1 = si.highpass_filter(raw_rec, freq_min=400.)
bad_channel_ids, channel_labels = si.detect_bad_channels(rec1)
rec2 = rec1.remove_channels(bad_channel_ids)
print('bad_channel_ids', bad_channel_ids)
rec3 = si.phase_shift(rec2)
rec4 = si.common_reference(rec3, operator="median", reference="global")
processed_recording = si.concatenate_recordings([rec4])
processed_recording

# Perform Sorting (this will take forever)
sorting = si.run_sorter(sorting_algorithm, processed_recording, output_folder=base_folder / sorting_algorithm / 'sorting_output',
                        docker_image=False, verbose=True, remove_existing_folder= True, delete_container_files=False, **sortingParams)
```

alejoe91, 2024-03-13:
Hi Ali, 

Thanks for the feedback!

Do you use the same environment? 
To make sure Ks4 uses the GPU you can try this from the notebook:
```
import torch

print(torch.cuda.is_available())
```

If the print is false, then you probably need to reinstall torch with vida support in the env, as explained in the Kilosort README.

I'll investigate what's wrong with the container. Haven't tested it on Windows, but it works on my side on Linux.

mohebial, 2024-03-13:
Thanks much for looking into this! I am running them both in the same environment (python=3.9, spikeinterface=1.0.1, KS=4.0.1). And the CUDA is available in the notebook. You will see below that it passes that test, and the docker image is running, but my GPU is not in use. The same happens for docker_image=False. But when I use straight KS4 to sort my binary data directly, it's very fast, and my GPU usage is 100%

Best,
Ali

[image]

alejoe91, 2024-03-13:
@zm711 pinging you here since you might be trying KS4 ok Windows too! Just wanted to know if you can replicate the issue

alejoe91, 2024-03-13:
@mohebial can you try this:
```
device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
print(device)
```

zm711, 2024-03-13:
Hey both. I was actually going to open an issue later today. CUDA is available on my device. When I was running the SpikeInterface wrapper it said it was going to take between 48-120 hours to run KS4 based on the constantly changing tqdm bar. So I thought maybe it was two things:

* the wrapper is slowing things down somehow
* the lack of binary file is causing slow reading

so I also tried to run KS4 after writing a binary file separately and loading into KS4 the binary file and the probe map. When I left at the end of the day yesterday it was running in my shell (no tqdm appeared for native KS4) so I was going to make a report after I got to lab to see if it had worked overnight or not. Based on the task manager it seems like it was still just using cpu instead. For me I tried to push it and use python 3.11 so I might downgrade to 3.10/3.9 and see if that’s part of my issue. But on their read me it says that CUDA issues might require more personal CUDA troubleshooting (e.g. my GPU is old and only has 4gb of dedicated memory and they rec 6 or 8 gb so maybe that is part of my problem). 

To summarize:
KS4 has yet to work for me with SI or without SI. But I’ll do an update to this when I get to lab.

zm711, 2024-03-13:
It crashed my computer overnight... I'll try again today and report back when I can. What I can say though is that with SI it was working. I'll try native KS4 one more time if that fails I'll play with the wrapper locally and add some print statements to see if the wrapper is accidentally doing something different for windows. 
As we can see the device is correct in the environment...
[image]

LIVE EDIT: Got it running, but it failed for me saying it could not find any data worth sorting (this dataset previously worked with KS3 and KS2 and MS5). I'll start at least poking around with the wrapper. But we can see that native KS4 finds the GPU:

[image]

One other strange note was that if I specify the dtype as np.int16 put specifying as 'int16' at least gets to the following error: 
[image]

So I'll poke around with our SI wrapper locally.

zm711, 2024-03-13:
https://github.com/MouseLand/Kilosort/blob/6c8ce1fb2362f23cc719943e00fef161b1d54bf5/kilosort/spikedetect.py#L48-L77

mohebial, 2024-03-13:
Thank you both! 
@alejoe91  In response to 
_device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
print(device)_
 I get _'cuda'_. 

@zm711 Everything takes longer. Here's a screenshot. It's been running since last night. Drift computation takes ~7hr and template extraction seems to take about the same... It is interesting how my GPU utilization is spiking (and the Memory is choking). 
[image]

zm711, 2024-03-13:
So based on my reading the drift correction and the actual sorting run (some of ) the same functions internally so I think that is where we are both getting stuck at. I didn't make it past drift correction, but I just skipped drift correction with the SI wrapper and I'm still having super slow speeds. My system memory is also nearly maxing out.

How did you run KS4 without spikeinterface? Did you write a binary file or use their `io.binary_from_spikeinterface`?

mohebial, 2024-03-13:
I don't think it's just the drift correction. I am now extracting templates, it's been over two hours and only 30% completed (npx1.0, 90min recording). To run the standalone KS4, I used their GUI and loaded my .dat file. It just worked fine, engaged the GPU and completed the sorting rapidly... @zm711

zm711, 2024-03-13:
> I don't think it's just the drift correction. I am now extracting templates, it's been over two hours and only 30% completed

yep that's what I was saying too. 

@alejoe91 I'm wondering if this does have to do with the lack of writing a binary file is slowing things down. I don't know why it would be fine on linux but not on Windows though. But kilosort doesn't have the same multiprocesssing for some of the io stuff, so maybe we are having an OS specific issue. 

For reference I chose a small dataset for testing KS4 (64 channel, 15 gb total) and my GPU, SSD, and CPU are all older than @mohebial so I'm getting even longer times for my steps. Like I said if I write a binary file from spikeinterface and then do `run_kilosort` it fails very quickly as it doesn't detect enough spikes.

zm711, 2024-03-15:
@mohebial, a PR was just merged to help engage the GPU in docker. So if you want to retry KS4 in your docker container you can check if your GPU is now being used. (Still haven't figured out a fix for local quite yet--from my end). I found out the reason why my dataset failed (but not why it was going so slow). KS4 cannot currently handle multishank probes.

mohebial, 2024-03-15:
Thanks @zm711. I will check and report back!

mohebial, 2024-03-16:
@ zm711 Now my sorting gets stuck in an infinite loop without any progress or errors... This might be an NVIDIA container toolkit issue and I don't know how to fix it. Will keep trying different things

zm711, 2024-03-17:
I don't really know docker so if Alessio has tips he'll let you know when he checks this thread.

mohebial, 2024-03-18:
Thanks! I've decided to switch to Ubuntu for spikeSorting, that might solve some of my issues...

zm711, 2024-03-18:
Sounds good. Feel free to open new issues as they come up :)

closed 2024-03-18 by mohebial