SpikeInterface/spikeinterface #2587: Handling sparsity for sorting_analyzer

OlivierPeron, 2024-03-15:
Hi everyone, 

I have been updating my code to use the new **SortingAnalyzer** class, and I have some questions about it. 

- First, I noticed that it's not possible to define sparsity using the 'by_property' method while creating an instance of the SortingAnalyzer class. In this scenario, sparsity can be defined later, applied to the instance, and then saved. Is this the only method, or is there a more straightforward approach?

  ```
  sparsity = si.compute_sparsity(sorting_analyzer,
                                 method = 'by_property',
                                 peak_sign = 'both',
                                 by_property = 'sh')
  
  sorting_analyzer.sparsity = sparsity
  
  sorting_analyzer.save_as(format = 'binary_folder',
                           folder = sa_path)
  ```

- I managed to save this sparsified instance in a new folder but not as an update to the original folder. Is there any way to update it?

- I also noticed that the properties from the recording and the sorting aren't inherited by the SortingAnalyzer instance. I assume that I should then set the properties as follows:

    `sorting_analyzer.recording.set_property('sh', group_rec)`
    `sorting_analyzer.sorting.set_property('sh', group_sort)`

Thanks !!

samuelgarcia, 2024-03-15:
Hi,
You cannot change the sparsity of a sorting analyzer, you need to start a new one.
So
```python
# you need a dense (because it is fast to create it!!!)
analyzer_dense = create_sorting_analyzer(sorting, recording, sparse=False)
sparsity = si.compute_sparsity(analyzer_dense, method = 'by_property', by_property = 'sh')

analyzer_sparse = create_sorting_analyzer(sorting, recording, sparsity=sparsity)
 # And then then make some computattion
analyzer_sparse.compute(...)
# and then save
analyzer_sparse.save_as(..)
```

And the property, you are setting like you do must be done done before otherwiseit  will not be perstistant on disk!
This is important.

OlivierPeron, 2024-03-15:
Ok thank you ! That clarifies it ! Just to be sure about the properties, you mean I should save the sorting on disk again after setting the properties ? For exemple : 
```
sorting.set_property('sh', group_sort)
sorting.save(args)
```

alejoe91, 2024-03-15:
We should have an option, but that's not in place yet!

closed 2024-04-10 by OlivierPeron