SpikeInterface/spikeinterface #2589: Waveform processing by unit very slow in version 0.101.0

taningh86, 2024-03-15:
Hi, 
Once the waveform figure is plotted its taking a lot of time (~2 mins) to process from waveform of one unit to another. This also happened with 0.100.0 but ``templates_percentile_shading=None`` got rid of the issue. With version 0.101.0 ``templates_percentile_shading=None`` don't seem to improve. Below is the code I used to compute amplitudes and get the plot

``sparsity = si.compute_sparsity(sorting_analyzer, method = "radius", radius_um = 30)
w = sw.plot_unit_templates(sorting_analyzer,templates_percentile_shading=None,backend = "ipywidgets")``

Is there another alternative with version 0.101.0? or is the issue soething other than the version?

Thanks
Jimmy

samuelgarcia, 2024-03-15:
Hi.
Maybe your sorting is dense and you want to have sparse one.
compute_sparsity, like uou do here, do means that you turn sorting_analyzer into a sparse one.

zm711, 2024-03-15:
Hey Sam here is the correctly rendered code from the original issue:

```python
from spikeinterface import create_sorting_analyzer, load_sorting_analyzer
sorting_analyzer = si.create_sorting_analyzer( sorting=sorting_wout_excess_spikes,  recording=recording_saved,  format='binary_folder',  folder= 'sorting_analyzer' )
job_kwargs = dict(n_jobs=8, chunk_duration="1s", progress_bar=True)
sorting_analyzer.compute( "random_spikes", method="uniform", max_spikes_per_unit=500, ) 
sorting_analyzer.compute("waveforms", ms_before=1.0, ms_after=2.0, return_scaled=True, **job_kwargs) sorting_analyzer.compute("templates", operators=["average", "median", "std"])
print(sorting_analyzer)
sparsity = si.compute_sparsity(sorting_analyzer, method = "radius", radius_um = 30) 
w = sw.plot_unit_templates(sorting_analyzer,templates_percentile_shading=None,backend = "ipywidgets")
```

The analyzer is already sparse at creation, so it shouldn't be a density issue. But yeah calculating the sparsity again doesn't do anything, but it also shouldn't hurt anything....

taningh86, 2024-03-16:
Sorry, its about 13 seconds of lag to toggle between units not 2 mins. I overshot it by a lot. However, its not instantaneous like how it used to be. Are there any other arguments I can change to make the process faster?

alejoe91, 2024-03-16:
@samuelgarcia I think we should pre-load the templates within the widget, especially when using ipywidgets backend!

samuelgarcia, 2024-03-18:
SAlut,
Normally when the templates extension is loaded then the array living in memory and so it is loaded for always until the analyzer is desoyed. So the access should be fast.
I will try to reproduce the lag when interactive plot template locally.

alejoe91, 2024-03-18:
@taningh86 can you check if this slowdown is only happening the first time you plot it? Because in that case, we know how and where to fix it :)

taningh86, 2024-03-18:
@alejoe91 Yeah, i did that too but don't improve processing speed. Is it better on your end?

taningh86, 2024-03-19:
Please let me know if there are any updates on this issue. Thanks!