SpikeInterface/spikeinterface #2596: Is it possible to access original data through preprocesed `BaseRecording` instance

rat-h, 2024-03-18:
The pipeline
```
Binary File -> Preprocessing -(1)-> Sorting -(2)-> Wave Extractor -(3)-> Neuron feature assessment. 
```
The last class gets `BaseRecording` after (1) and uses `get_traces` to compute a spike "fingerprint". Unfortunately, preprocessing removes something we are looking for. Would it be possible to `get_traces` from the original Binary File without reloading it and creating an additional `BaseRecording` object?

In other words, is there something like that
```python
# preproc is BaseRecording (1)
# which is pointing to the filtered and zeroed data
fingerprint = array([
        preproc.original_data.get_traces(
            start_frame   = sp+frmstart,
            end_frame     = sp+frmend).T
        for sp in sp_train
        ])
```

alejoe91, 2024-03-18:
Hi @rat-h 

This is something that is not implemented yet, but I agree would be super useful.

What about having a `recording/sorting.get_parent()`? This could be automatically implemented by all preprocessors/slicing/ etc.

samuelgarcia, 2024-03-18:
Sometimes it is several parents!

rat-h, 2024-03-18:
@alejoe91 @samuelgarcia That is exactly what I was looking for! Perfect! Thank you :)!

alejoe91, 2024-03-21:
@rat-h this is not implemented yet. Just made a PR!

Note: this only works with parents that are single objects. The `concatenate_recordings`/`aggregate_sortings`, etc. will not have this functionality

PR #2610 (2024-03-29): Implement `get_parent()` function
Fixes #2596

This is super useful for example to plot different processing layers!
closed 2024-03-29 by samuelgarcia