SpikeInterface/spikeinterface #2599: compare_multiple_templates error

AOONIH, 2024-03-18:
Hi,

I'm trying to compare sorter outputs over days and tried using the compare_multiple_templates function.  
I get this error:
ValueError: Shape of passed values is (191, 191), indices imply (191, 132)
which comes from: `self.agreement_scores = pd.DataFrame(agreement_scores, index=self.unit_ids[0], columns=self.unit_ids[1])`
I'm not sure why agreement_scores is a square matrix if the number of units is different across the 2 sorters?

AOONIH, 2024-03-18:
I think I found the issue:
Line 115 in template_similarity.py specifies the function postion arguments incorrectly:
`_compute_template_similarity(waveform_extractor, waveform_extractor_other, method)`

zm711, 2024-03-19:
Fixed by #2601.

closed 2024-03-19 by zm711