SpikeInterface/spikeinterface #2615: MaxWell Extractor Error - cannot unpack non-iterable Extractor Object

wesleypclawson, 2024-03-21:
Hi!

 It may be an issue for neo, but trying to open a recording from a MaxOne or MaxTwo (6 well) results in a TypeError. I have the code posted below for MaxOne, but it's the same for MaxTwo (however with a MaxTwo there's an additional stream_id/stream_name variable passed). 

I have the current decompression already installed on the machine as well. 

**Code:** 
```
data_location = '/home/mxwbio/DataBackup/MaxOne/000095/'
data_name = 'data.raw.h5'
recording, sorting_true = se.read_maxwell(data_location + data_name)
print(recording)
print(sorting_true)
```

**Error:**
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[51], line 3
      1 data_location = '[/home/mxwbio/DataBackup/MaxOne/000095/](http://localhost:8888/lab/tree/sitest/DataBackup/MaxOne/000095/)'
      2 data_name = 'data.raw.h5'
----> 3 recording, sorting_true = se.read_maxwell(data_location + data_name)
      4 print(recording)
      5 print(sorting_true)

TypeError: cannot unpack non-iterable MaxwellRecordingExtractor object
```

alejoe91, 2024-03-21:
Hey @wesleypclawson 

The read function only returns a recording! Si you have to remove `, sorting_true`!

wesleypclawson, 2024-03-22:
Ohhhhh, I see now. This is why we don't try new things after long days, hah. 

Working now!

closed 2024-03-22 by alejoe91