SpikeInterface/spikeinterface #2618: Trouble with Maxwell HDF5 Compression

wesleypclawson, 2024-03-22:
Hi again, 

 When trying to run any of the external spike sorters via docker, I get errors thrown that there's no decompression algorithm and the sorting fails. It works if I run things outside of docker, so I assume I need to make a custom dockerfile that copies the current decompression over before running the spike sorting? or is there a simpler fix?

What is the suggested 'best practices' path forward?

**Code:**
```
sorting = ss.run_sorter(sorter_name='kilosort2',
    recording=recording,
    output_folder="Sorting/kilosort2",
    docker_image=True)
```

**Error:**
```
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[21], line 1
----> 1 sorting = ss.run_sorter(sorter_name='kilosort2',
      2     recording=recording,
      3     output_folder="Sorting[/kilosort2](http://localhost:8888/kilosort2)",
      4     docker_image=True)
      6 print(sorting)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:169, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    167         else:
    168             container_image = singularity_image
--> 169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
    175 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:595, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, **sorter_params)
    593 if run_error:
    594     if raise_error:
--> 595         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    596 else:
    597     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
**********
Maxwell file format is based on HDF5.
The internal compression requires a custom plugin!!!
This is a big pain for the end user.
You, as a end user, should ask Maxwell company to change this.
Please visit this page and install the missing decompression libraries:
https://share.mxwbio.com/d/4742248b2e674a85be97/
Then, link the decompression library by setting the `HDF5_PLUGIN_PATH` to your
installation location, e.g. via
os.environ['HDF5_PLUGIN_PATH'] = '/path/to/custom/hdf5/plugin/'

Alternatively, you can use the auto_install_maxwell_hdf5_compression_plugin() below
function that do it automagically.

**********
```
The above was repeated a number of times, followed by:

```
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/mxwbio/sitest/Sorting/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "[/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 224](http://localhost:8888/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py#line=223), in run_sorter_local
    SorterClass.setup_recording(recording, output_folder, verbose=verbose)
  File "[/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 230](http://localhost:8888/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py#line=229), in setup_recording
    cls._setup_recording(recording, sorter_output_folder, sorter_params, verbose)
  File "[/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py", line 155](http://localhost:8888/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py#line=154), in _setup_recording
    write_binary_recording(
  File "[/root/.local/lib/python3.9/site-packages/spikeinterface/core/recording_tools.py", line 137](http://localhost:8888/root/.local/lib/python3.9/site-packages/spikeinterface/core/recording_tools.py#line=136), in write_binary_recording
    executor.run()
  File "[/root/.local/lib/python3.9/site-packages/spikeinterface/core/job_tools.py", line 401](http://localhost:8888/root/.local/lib/python3.9/site-packages/spikeinterface/core/job_tools.py#line=400), in run
    for res in results:
  File "[/home/miniconda3/lib/python3.9/concurrent/futures/process.py", line 562](http://localhost:8888/home/miniconda3/lib/python3.9/concurrent/futures/process.py#line=561), in _chain_from_iterable_of_lists
    for element in iterable:
  File "[/home/miniconda3/lib/python3.9/concurrent/futures/_base.py", line 609](http://localhost:8888/home/miniconda3/lib/python3.9/concurrent/futures/_base.py#line=608), in result_iterator
    yield fs.pop().result()
  File "[/home/miniconda3/lib/python3.9/concurrent/futures/_base.py", line 446](http://localhost:8888/home/miniconda3/lib/python3.9/concurrent/futures/_base.py#line=445), in result
    return self.__get_result()
  File "[/home/miniconda3/lib/python3.9/concurrent/futures/_base.py", line 391](http://localhost:8888/home/miniconda3/lib/python3.9/concurrent/futures/_base.py#line=390), in __get_result
    raise self._exception
OSError: Can't synchronously read data (can't open directory ([/usr/local/hdf5/lib/plugin](http://localhost:8888/usr/local/hdf5/lib/plugin)). Please verify its existence)
```

alejoe91, 2024-03-22:
Yeah this is tricky. I recommend getting rid of the Maxwell data format and converting to a binary folder as a first step:
```
recording_binary = recording.save(folder="wher-you-wanto-to-save-it", n_jobs=-1)
```
This should also speed up every other step :)

wesleypclawson, 2024-03-22:
Ah that makes tons of sense. I was thinking it'd be easier to do it the other way around. 

Works - now I'm running into out of memory issues (30 min recording / 1020 channels). I have 251 GB of RAM, so shouldn't be an issue there. I'll poke around the docs to find some parameters I can try changing. 

```
{
    "sorter_name": "kilosort2",
    "sorter_version": "compiled",
    "datetime": "2024-03-22T16:35:12.727259",
    "runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Determining good channels..",
        "----------------------------------------Out of memory."
    ],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 258, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py\", line 217, in _run_from_folder\n    raise Exception(f\"{cls.sorter_name} returned a non-zero exit code\")\nException: kilosort2 returned a non-zero exit code\n",
    "run_time": null
}
```

alejoe91, 2024-03-22:
That's GPU RAM!

wesleypclawson, 2024-03-22:
Is it? When I run `nvidia-smi` it shown just 4 Gb. If I run `free -m` I get 257369., so I assume it's a 4GB GPU and 251 system RAM. 

However, I guess if I enabled docker to use GPU compute, is the out of memory coming from the GPU?

alejoe91, 2024-03-22:
I think so..4RAM on the GPU might be too low for running KS, especially for such large recording

wesleypclawson, 2024-03-22:
Sadly I problem I can't fix at the moment. Any suggestions for a sorter of similar quality that relies more on CPU?

naterenegar, 2024-03-22:
HDsort was specifically designed on maxwell meas, and runs on CPU

wesleypclawson, 2024-03-22:
Will try! @alejoe91 I think we can close this as the original issue was solved. Thanks

alejoe91, 2024-03-22:
You could also try SpykingCircus2, which is brand new and fully spikeinterface-based. HDsort is going to be very slow and with mixed results...

closed 2024-03-22 by alejoe91

wesleypclawson, 2024-03-22:
@alejoe91 - do any issues using spyking circus go on this issue board or on spyking circus' github? 

error in question -> `ValueError: polyorder must be less than window_length.` during feature extraction

alejoe91, 2024-03-22:
For spyking circus 2 you can use this repo