SpikeInterface/spikeinterface #2626: A thread to discuss inter-session alignment.

JoeZiminski, 2024-03-26:
Following a discussion on the Slack channel about inter-session alignment I thought I'd start an issue to summarise / discuss these things and come up with a plan on how this could be handled in Spikeinterface.

There are a couple of (not necessarily mutually exclusive) approaches:

**1) post-hoc unit-matching**
This approach matches units from independently sorted sessions. Current approaches includes Unitmatch(https://www.biorxiv.org/content/10.1101/2023.10.12.562040v1.full.pdf). SI and Unitmatch teams are working together to ensure compatibility such that SI-sorted data can be easily read in Unitmatch. There is also the approach outlined in [this paper](https://elifesciences.org/reviewed-preprints/92495) which looks interesting. Typically the motion correction in these methods relies on putative matches between units across sessions, which may fail in the presence of large (e.g. >50 um?) drift. Therefore alignment of either the traces (pre-sorting) or templates (post-sorting) across sessions, would compliment these methods.

**2) Concatenating sessions and motion correcting**
Another approach is to concatenate two sessions together and perform motion correction as if they were a single session, then sort. A concern is how current intra-session methods will perform in the presence of significant drift between sessions (manifesting as a sudden, large 'jump' in the middle of the concatenated recording). In the NP2 paper the Kilosort team suggest this approach does not work so well with the iterative template method*. In the [Dredge paper](https://www.biorxiv.org/content/10.1101/2023.10.24.563768v1) this approach appears to be used successfully. It would be interesting to see at what level of inter-session drift (linear and nonlinear) this approach fails for the available methods. I guess the machinery for such tests has already been largely developed [here](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC10897502/). 

**3) Alignment of raw data pre-sorting**
This approach is described in the NP2 paper, roughly it goes like 1) within-session motion correction for two sessions 2) compute drift between session and add compensation to within-session offsets 3) interpolate. In theory it should also be quite possible to do with the existing `correct_motion` machinery, just requiring some additional logical to incorporate inter-session registration offsets prior to interpolation. 

**4) Alignment of template waveforms post-sorting**
This approach is related to **post-hoc unit matching (1)** specifically. For these methods, instead of interpolating the raw data based on estimated inter-session drift prior to sorting, only the templates are shifted prior to unit-matching. (@alejoe91 I think this is the idea you had at the Unitmatch meeting, if I understood correctly). This should have the same effect as (3) for unit-matching but would be much faster.

In terms of action-points, to my knowledge (1) is actively being worked on but (3) and (4), as well as validation discussed in (2) are not. Therefore supporting these inter-session approaches in Spikeinterface might be a nice idea if others think they have utility.

* _Kilosort team suggest this approach does not work well_
```
Chronic recordings
For chronic recordings, we tried to concatenate the raw data files and run the Datashift algorithm as
described above. The results were not always satisfactory, especially when the recording sessions were
separated by weeks.
``` from [here](https://www.biorxiv.org/content/10.1101/2020.10.27.358291v1.full.pdf).

samuelgarcia, 2024-03-29:
And this could be a good project for the next [hackthon in Edinburgh](https://spikeinterface.github.io/spikeinterface-events/spikeinterface-workshop-2024/)

florian6973, 2024-06-04:
Very interesting thread!

I think https://github.com/SpikeInterface/spikeinterface/issues/2911 could be related to 1) if we try to compute the best unit matching possible across all sessions based on some kind of similarity metrics.

JoeZiminski, 2024-06-27:
Thanks everyone, I was thinking to get started on this soon. 

1) **Estimate Alignment Across Sessions**

The API here is inspired by the existing within-session `si.correct_motion()`. This function takes an ordered set of recordings as input. Almost always they would already be motion corrected within-session. The function will compute the vertical displacement across sessions using existing methods. In the first instance this can be fairly straightforward - use some random chunks from the recordings to estimate the activity histogram for each session, and optimise to shift them into a middle-position with minimum loss of top/bottom of the probe data and keep similar interpolation distances per session. In future this could incorporate LFP band also.

This function outputs a list of corrected motion recordings using the existing `InterpolateMotionRecording`. Whats super nice about the existing SpikeInterface API is that the motion correction is lazy. So if motion correction is already performed, all that needs to be done is for the estimated inter-session drift correction to be added to existing drift correction in the motion-correction object.

The inter-session correct displacement function can also output a `motion_info` object similar to `si.correct_motion()`. This can then be used to correct just the peaks, or directly the templates (next step). @samuelgarcia @cwindolf it would be great to get your feedback on this.

2) **Motion correct templates before unit matching**

It may be that for some cases interpolating the raw data is too heavy. Instead Alessio suggested interpolating only the templates before unit matching. Using the output `motion_info` as above, this could be passed to a `Sorting` (?) object function that shifts the peak and template locations only based on the inter-session motion. This should help a lot when unit-matching based on templates when there is a lot of inter-session drift.

3) **unit matching** 

There is already some template-based unit matching [in SpikeInterface](https://spikeinterface.readthedocs.io/en/latest/modules/comparison.html#template-comparison) and we have been working with UnitMatch who have kindly implemented a SpikeInterface data loader. So we could possibly merge these into some kind of `InterSessionMatching` object to perform the matching while exposing different backends. 

Would be great to get feedback on this! I

florian6973, 2024-07-02:
Sounds like a great plan! I am less familiar with motion correction but definitely glad to help for the spikeinterface template-based unit matching.

I would strongly suggest that alignment be an option so that we can 'skip' it when it is not needed. Ideally I would also like to be able to convert the Matching object to a Sorting one to postprocess it as usual with an (extended?) sorting (matching?) analyzer. And we should definitely implement some quality metrics on how good the unit matching is, I do not know if you have already some in mind.

JoeZiminski, 2024-07-09:
Thanks @samuelgarcia, that's okay, cool to hear about the future plans for including segment! For now each recording could just be handled as a separate session, and in future this extended to the multi-segment case along with the motion estimation?

JoeZiminski, 2024-08-20:
Another approach of interest [here](https://github.com/AugustineY07/Neuron_Tracking)

JoeZiminski, 2024-10-15:
relevant: [here](https://github.com/MouseLand/Kilosort/issues/388)

samuelgarcia, 2025-08-20:
Charlie, I propose something :  you offer me a flight ticket to new york and then we code this togoether!