SpikeInterface/spikeinterface #2634: Key error when loading extension "amplitude_scalings"

Antho2422, 2024-03-27:
Hi, 

I get this key error on "amplitude_scaling" when trying to load a saved SortingAnalyzer with extensions. 
```
--> sortinganalyzer.py:139: return SortingAnalyzer.load(folder, load_extensions=load_extensions, format=format)

--> sortinganalyzer.py:257: sorting_analyzer.load_all_saved_extension()

--> sortinganalyzer.py:993: self.load_extension(extension_name)

--> sortinganalyzer.py:978: extension_class = get_extension_class(extension_name)

-> sortinganalyzer.py:1112: ext_class = extensions_dict[extension_name]

KeyError: 'amplitude_scalings'
``` 

I cannot figure out what is the issue... It is probably happening too with all the other extensions I guess

If somebody knows I would be grateful !

Thanks, 
Anthony

alejoe91, 2024-03-27:
Hi @Antho2422 

You previously computed the `amplitude_scalings` extension, right?

Antho2422, 2024-03-27:
Yes, I also have the corresponding folder 

[image]

zm711, 2024-03-27:
This happened to me once when I updated the code mid analysis. Restarting my kernel and then doing the loading worked for me. The dictionary is made at first import and isn't updated so I think that causes this issue.

Antho2422, 2024-03-27:
@zm711 It worked, thank you for the suggestion. I did not think this could be the issue

zm711, 2024-03-27:
Not sure how to fix this issue or if it is fixable with the current codebase, but yeah if you get errors like that I would recommend just restarting the kernel while we keep trying to polish the analyzer code.

alejoe91, 2024-03-27:
Yeah I don't think this can be fixed, since it's a problem (?) of the registration of extensions.

Thansk Zach for figuring this out!

closed 2024-03-27 by alejoe91