SpikeInterface/spikeinterface #2638: Installation error: 'create_sorting_analyzer' missing

mtehrani-code, 2024-03-27:
Hello,

I am new to SI and Python. I am trying to download SI on my machine and follow the steps outlined here for quick installation:
https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/README.md

I have the full_spikeinterface_environment_windows.yml saved in my documents. However, when I check the installation, I get these error messages:

Run tridesclous ...Fail, Error: module 'spikeinterface.full' has no attribute 'create_sorting_analyzer'
and 
Open spikeinterface-gui ...Fail, Error: module 'spikeinterface.full' has no attribute 'load_sorting_analyzer'
Export to phy ...Fail, Error: module 'spikeinterface.full' has no attribute 'load_sorting_analyzer'

It looks like the sorting_analyzer method might be missing from the code?

Could you please help me fix this issue?

Thanks!

Mahtab

zm711, 2024-03-28:
Actually, I think Sam updated the test_your_install for 0.101.0, but the yaml still installs 0.100.1/0.100.2. So that's the disconnect. This will be fixed with the 0.101.0 release. Or do we want to add back the old check_your_install instead?

mtehrani-code, 2024-03-28:
@alejoe91  

This is the link I followed from within the latest release documentation (quick installation for beginners):

https://github.com/SpikeInterface/spikeinterface/tree/main/installation_tips

After downloading Anaconda and the full_spikeinterface_environment_windows.yml, I checked the installation using the code: 

conda activate si_env
python check_your_install.py

Then I get the error messages.

M

mtehrani-code, 2024-03-28:
> Actually, I think Sam updated the test_your_install for 0.101.0, but the yaml still installs 0.100.1/0.100.2. So that's the disconnect. This will be fixed with the 0.101.0 release. Or do we want to add back the old check_your_install instead?

This makes sense...How do you recommend that I proceed?

M

samuelgarcia, 2024-03-29:
Hi.
Indeed, As Zach mentioned, the `check_your_install.py` is testing the future release so need a installation from soure.
Sorry that.

mtehrani-code, 2024-03-29:
> You could test the rolling install (the other yaml [full_spikeinterface_environment_rolling_updates.yml](https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/full_spikeinterface_environment_rolling_updates.yml) which will automatically download from source. Or we could explain how to download from source. It's actually not too bad even if you're a beginner :)
> 
> Technically you don't need to do the `check_your_install`. So the other option would be you could just use your current environment and follow tutorials in [these docs](https://spikeinterface.readthedocs.io/en/0.100.1/) that Alessio shared.
> 
> How would you like to proceed? Happy to help you either way. The new SortingAnalyzer is what will work going forward so it might be best to just get used to it now.

Thank you! I'm now trying to download from source. Trying to follow the documentation you have provided...but not getting very far. I would appreciate if you could walk me through it in a very beginner friendly way :D

zm711, 2024-03-29:
Yeah of course. So if you installed with the yaml I'm assuming you have conda. Do you have Github desktop app or are you using git? Basically to install from source you could clone the repo. You've got a few options. zip is less idea because it won't update as we update so ideally you clone or use github desktop or you can use a git client to clone yourself if you don't have the desktop app.

[image]

We need to address one thing really quick are you a Windows user or a Mac/Linux user? It is customary to represent the terminal for mac/linux with a `$` so you would never type the `$` you would just type what comes after. If you are a windows users then instead you will see a `>`. So wherever I put a `$` just replace with a `>` in your head. Also to note if you're a Windows user you will need to open the Anaconda prompt whereas on Mac (and I think linux) you just open the terminal. Finally the `(base)` lets you know which environment you are in. For Mac it appears on the terminal line. For Windows it will also appear before the `>` but only if you open the Anaconda Prompt. If you accidentally open the Command Prompt it won't be there.

Once we have the repo downloaded then we type the following
```
(base) $ cd documents/github/spikeinterface
```

If you installed your github folder somewhere else or you decided to just use a zip file then you need to type `cd` to wherever your file is.
Then we would type:

```
(base) $ conda create -n spikeinterface_env -c conda-forge python=3.10
```
The `-n` flag is the name you want to give to the environment. You can choose whatever you want (like si_env or whatever). `-c` is the channel where you want to pull things from. Honestly for this install it doesn't really matter, but in general using `conda-forge` gives you the most up to date packages whereas default gives you the most stable packages. Our testing is done on python 3.10 so it's probably safest to use python 3.10 (I use 3.11 with no problem though). Which IDE/coding environment do you plan to use. If you are a spyder user rather than a jupyter user then it might be better to type

```
(base) $ conda create -n spikeinterface_env -c conda-forge python=3.10 spyder
```

It will ask if you want to install the necessary packages type `y`.

Then
```
(base) $ conda active spikeinterface_env
```
This activates your conda environment. You should always install into an environment rather than base (just to protect the base install). Once you confirm you're in the correct environment you type

```
(spikeinterface_env) $ pip install -e .[full,widgets]
```

The `-e` means editable mode which allows you to pull changes and incorporate them into the environment. The `.` means create the package that is in the folder where I currently am (and since we moved into the spikeinterface folder earlier it will build there).  `[full,widgets]` will ensure you download all the extra packages necessary for the full functionality of spikeinterface including all of its plotting functionality. 

Now if you want to test something like mountainsort5 you just do:

```
(spikeinterface_env) $ pip install mountainsort5
```

mtehrani-code, 2024-04-01:
> Yeah of course. So if you installed with the yaml I'm assuming you have conda. Do you have Github desktop app or are you using git? Basically to install from source you could clone the repo. You've got a few options. zip is less idea because it won't update as we update so ideally you clone or use github desktop or you can use a git client to clone yourself if you don't have the desktop app.
> 
> ![image](https://private-user-images.githubusercontent.com/92116279/318124757-13870cb7-40ff-4990-a7e9-546a76fd5ec1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5OTAzMzEsIm5iZiI6MTcxMTk5MDAzMSwicGF0aCI6Ii85MjExNjI3OS8zMTgxMjQ3NTctMTM4NzBjYjctNDBmZi00OTkwLWE3ZTktNTQ2YTc2ZmQ1ZWMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDE2NDcxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRiOTBhNTVjZmY0MDFlMDM5YzgzM2NkMDFhODlmOGY5YjJlZjQwNWM3N2NiYzZjMzlmMGQzMjExOWNlMmQ4ZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nhVFuVjTx1VHbWFZQKfXuWv217zWwBbk2nqTit--16U)
> 
> We need to address one thing really quick are you a Windows user or a Mac/Linux user? It is customary to represent the terminal for mac/linux with a `$` so you would never type the `$` you would just type what comes after. If you are a windows users then instead you will see a `>`. So wherever I put a `$` just replace with a `>` in your head. Also to note if you're a Windows user you will need to open the Anaconda prompt whereas on Mac (and I think linux) you just open the terminal. Finally the `(base)` lets you know which environment you are in. For Mac it appears on the terminal line. For Windows it will also appear before the `>` but only if you open the Anaconda Prompt. If you accidentally open the Command Prompt it won't be there.
> 
> Once we have the repo downloaded then we type the following
> 
> ```
> (base) $ cd documents/github/spikeinterface
> ```
> 
> If you installed your github folder somewhere else or you decided to just use a zip file then you need to type `cd` to wherever your folder is. Then we would type:
> 
> ```
> (base) $ conda create -n spikeinterface_env -c conda-forge python=3.10
> ```
> 
> The `-n` flag is the name you want to give to the environment. You can choose whatever you want (like si_env or whatever). `-c` is the channel where you want to pull things from. Honestly for this install it doesn't really matter, but in general using `conda-forge` gives you the most up to date packages whereas default gives you the most stable packages. Our testing is done on python 3.10 so it's probably safest to use python 3.10 (I use 3.11 with no problem though). Which IDE/coding environment do you plan to use. If you are a spyder user rather than a jupyter user then it might be better to type
> 
> ```
> (base) $ conda create -n spikeinterface_env -c conda-forge python=3.10 spyder
> ```
> 
> It will ask if you want to install the necessary packages type `y`.
> 
> Then
> 
> ```
> (base) $ conda activate spikeinterface_env
> ```
> 
> This activates your conda environment. You should always install into an environment rather than base (just to protect the base install). Once you confirm you're in the correct environment you type
> 
> ```
> (spikeinterface_env) $ pip install -e .[full,widgets]
> ```
> 
> The `-e` means editable mode which allows you to pull changes and incorporate them into the environment. The `.` means create the package that is in the folder where I currently am (and since we moved into the spikeinterface folder earlier it will build there). `[full,widgets]` will ensure you download all the extra packages necessary for the full functionality of spikeinterface including all of its plotting functionality.
> 
> Now if you want to test something like mountainsort5 you just do:
> 
> ```
> (spikeinterface_env) $ pip install mountainsort5
> ```
> 
> EDIT: I had a typo above (it's `activate` and not `active` for conda :) )

Thanks very much! This is very helpful. I am using windows, using anaconda, and I downloaded GitHub desktop.

OK. I got to that last step, but when I type in: (spikeinterface_env) $ pip install -e .[full,widgets]
I get this error:

(spikeinterface_env) C:\Users\matta>pip install -e .[full,widgets]
Obtaining file:///C:/Users/matta
ERROR: file:///C:/Users/matta does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

Any idea what this might be due to?

M

mtehrani-code, 2024-04-01:
> Thanks very much! This is very helpful. I am using windows, using anaconda, and I downloaded GitHub desktop.
>
> OK. I got to that last step, but when I type in: (spikeinterface_env) $ pip install -e [...]

I guess I shouldn't be in the C:\Users\matta> directory? Not sure how to get out of it...

mtehrani-code, 2024-04-01:
@zm711 Never mind! I hadn't cloned the repository correctly. I redid that and it seems to be working...

zm711, 2024-04-01:
Awesome! You can try the check_your_install (and then on windows you need to do cleanup_for_windows). And as long as that is good you're good to go. Feel free to open more issues as they come up :)

mtehrani-code, 2024-04-01:
> Awesome! You can try the check_your_install (and then on windows you need to do cleanup_for_windows). And as long as that is good you're good to go. Feel free to open more issues as they come up :)

When I run the python check_your_install.py, I'm getting a number of errors (see below):

(spikeinterface_env) C:\Users\matta\Documents>python check_your_install.py
write_binary_recording with n_jobs = 1 and chunk_size = 30000
write_binary_recording: 100%|########################################################| 200/200 [00:09<00:00, 21.75it/s]
Import spikeinterface ...OK
Import spikeinterface.full ...OK
Run tridesclous ...Fail, Error: The sorter tridesclous is not installed.Please install it with:

To use Tridesclous run:

       >>> pip install tridesclous

    More information on tridesclous at:
      * https://github.com/tridesclous/tridesclous
      * https://tridesclous.readthedocs.io

Run tridesclous2 ...Fail, Error: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\matta\Documents\GitHub\SpikeInterface\src\spikeinterface\sorters\basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\matta\Documents\GitHub\SpikeInterface\src\spikeinterface\sorters\internal\tridesclous2.py", line 108, in _run_from_folder
    import hdbscan
ModuleNotFoundError: No module named 'hdbscan'

Spike sorting failed. You can inspect the runtime trace in C:\Users\matta\Documents\sorter_with_tridesclous2/spikeinterface_log.json.
Open spikeinterface-gui ...Fail, Error: No module named 'spikeinterface_gui'
Export to phy ...Fail, Error: Waveform folder does not exists

Do I need to fix this somehow?

zm711, 2024-04-01:
Hdbscan does better with a conda install. You’re actually okay to go unless you were hoping to directly test tridesclous. Which sorter were you hoping to run?

mtehrani-code, 2024-04-01:
Oh, Okay. I'm honestly not 100% sure about the sorter at the moment...I'm hoping spikeinterface can help me figure out which sorter would work best on my data...I'll definitely want to test kilosort.

If you think I don't need to worry about the checkinstall at the moment I will move on to the next step :)

zm711, 2024-04-01:
Yep I would try moving on for now. One thing you’ll get used to with python is troubleshooting these package install issues. If you decide to test tridesclous in the future we can get hdbscan to install using conda (we may need to install it earlier). That is why the yaml file is nice. It installs everything at the beginning. But we can troubleshoot that in the future.

mtehrani-code, 2024-04-01:
Okay. Thank you.

Now with the windowns cleanup, when I type in:

(spikeinterface_env) C:\Users\matta\Documents>conda deactivate
(base) C:\Users\matta\Documents>python cleanup_for_windows.py

Nothing seems to be happening. I'm back to:

(base) C:\Users\matta\Documents>

I assume I need to exit the spikeinterface_env first, correct?

zm711, 2024-04-01:
The cleanup just deletes the test files. So it will seems like it’s doing nothing but it’s just making sure that there are no test files left behind.

mtehrani-code, 2024-04-01:
So I should now be importing the import spikeinterface.full as si, correct?

When I type this into the conda prompt it tells me that import is not recognized as an internal or external command...I did some googling and it looks like import is a python command? Meaning I need to work in a python interpreter environment? Sorry, if I'm asking about the obvious.

zm711, 2024-04-01:
No worries. We all have to start somewhere! The Anaconda prompt is not the same as being in a Python repl. What most people like is using an IDE (that’s why I suggested spyder above) or you should have access to Jupyter. So in the Prompt you can type:

```
python
```
For a Python repl. Or `ipython` for an iPython repl. Or `jupyter` to start a Jupyter session. And once you’re in one of those you can use import and other commands. Probably worth reading a bit to see which approach you want to take. Also there will be a spikesorting conference in Edinburgh in May that may be of interest to you where there will be a workshop on how to do all this stuff. Might be worth seeing if there is still space. (I’ll share the link when I’m back at my computer).

mtehrani-code, 2024-04-01:
@zm711 I got it to work from within Spyder. Following the steps in the tutorial now. Thank you very much for all your help!

M

zm711, 2024-04-01:
Awesome. I'll close this issue since it was related to installation, but if you run into new problems feel free to open new issues!

chiyu1203, 2024-05-09:
Hi! I just wanted to say I saw similar 'create_sorting_analyzer' missing when I installed spikeinterface via pip. Installing from source fixed the problem.

olivershetler, 2024-06-11:
@chiyu1203 I'm having the same issue with pip installing spikeinterface[full] in my virtual environment.

olivershetler, 2024-06-11:
Any chance someone can just go add the function to the __init__ and push a new release to PyPI? This feels like a pretty important thing to have broken in public releases, given the push to unify all the post-processing with this SortingAnalyzer.

alejoe91, 2024-06-12:
@olivershetler we're ironing out the final details

You can get a pre release with the `SortingAnalyzer` with
```
pip install --pre spikeinterface
```

bxy666666, 2024-06-25:
> @chiyu1203我在虚拟环境中使用 pip 安装 spikeinterface[full] 时遇到了同样的问题。

Hi，may I ask your solution？

chiyu1203, 2024-06-25:
@bxy666666  @olivershetler  The only thing I did to fix my problem is to install spikeinterface "From Source". I stopped using `pip install` already. Please follow this [tutorial](https://spikeinterface.readthedocs.io/en/latest/get_started/installation.html) for more details.