SpikeInterface/spikeinterface #2663: November 2023 Tutorial: AppLayout shows up incomplete in Jupyter Notebook

mtehrani-code, 2024-04-03:
Hello,

I'm using Jupyter notebook to run the tutorial in a new file. I'm at the step where I want to visualize traces after having applied the filter and common referencing. However, this is what I see in the tutorial for the AppLayout specification for plotting the traces:

AppLayout(children=(TimeSlider(children=(Dropdown(description='segment', options=(0,), value=0), Button(icon='…

It looks like it is cut-off.

Does this mean I'm possibly not using the correct version of the JupyterNotebook? If so, what version do you recommend I use? If not version related, any ideas why this is happening?

Thank you,

MT

zm711, 2024-04-03:
The text is cutoff or there are issues with the figures?

mtehrani-code, 2024-04-04:
> The text is cutoff or there are issues with the figures?
Text is cut off, but plot appears OK. Ignoring the cut-off text for now, since I don't think it's causing any problems. Please ignore if you like.

zm711, 2024-04-04:
Sweet. Sometimes there are config files for Juypter (rendering with the html can be a mess). So if the actual stuff we care about is still appearing fine I would ignore that type of thing. I'll close for now :) Reopen if needed.

closed 2024-04-04 by zm711