SpikeInterface/spikeinterface #2664: Spikes are appearing as flipped, or upside-down

TheAnton205, 2024-04-03:
Hello, plotting traces in spike interface seem to plot spikes going down, rather than up. Is it possible to flip the y-axis and make them not upside down?

Below is what I want to see:
[image]

Here is what spikeInterface is plotting:
[image]

zm711, 2024-04-03:
Extracellular potentials typically go down? What kind of data are you hoping to plot? Which extractor are you using?

TheAnton205, 2024-04-10:
We are using openephys for extracting and neuropixels are the type of data we want to plot. The first image is a screenshot from openephys during live recording.

zm711, 2024-04-10:
I would guess that you have some sort of display issue for openephys. Or a referencing issue. In general extracellular potentials are downward deflecting the way the plot traces are. If you have sample data you can share we could double check for you. But the `get_traces()` function shouldn't flip your data.

Do you expect positive data for any other reason other than that is how the gui is displaying it?

zm711, 2024-04-10:
Of course. Feel free to open more issues if they come up. Happy to help!