SpikeInterface/spikeinterface #2668: correlograms, How should they look like?

sofiapuvogelvittini, 2024-04-04:
Hi, 
I am trying to use the correlograms to estimate connectivity metrics, but I'm a bit puzzled by the patterns I'm observing when I plot them. Most of the time, the auto-correlograms appear to be flat, which is contrary to what I was expecting... Shouldn't I expect to see some peaks here?
Only for one neuron I observed peaks in the autocorrelogram (attached pictures)
[correlograms.pdf](https://github.com/SpikeInterface/spikeinterface/files/14865864/correlograms.pdf)

Could you assist me in interpreting the output of correlograms? What is the expected pattern, and how should they ideally appear? I haven't come across tutorials that focus on this aspect.
Thanks a lot

This is my function to plot:

def plot_correlograms(correlograms):
    num_neurons = correlograms.shape[0]
    num_bins = correlograms.shape[2]

    for i in range(num_neurons):
        for j in range(num_neurons):  # Iterate over pairs only once
            plt.figure()
            plt.plot(range(num_bins), correlograms[i, j], label=f'Neuron {i} vs Neuron {j} Correlogram')
            plt.xlabel('Time Bin')
            plt.ylabel('Correlation Strength')
            plt.title(f'Correlogram: Neuron {i} vs Neuron {j}')
            plt.legend()
            plt.show()

plot_correlograms(correlograms)

sofiapuvogelvittini, 2024-04-04:
Hi, thanks for your prompt reply!
I tried the internal function sw.plot_correlograms(analyzer), but I obtain the error: module 'spikeinterface.widgets'  has no attribute 'plot_correlograms' ... Also, I can not find the specific function plot_correlograms in the widgets.. maybe I do not understand very well where/how is this function located.

zm711, 2024-04-04:
That is because I had a typo. I'm sorry!

It's `sw.plot_crosscorrelograms`!

sofiapuvogelvittini, 2024-04-04:
Thanks! I tried the internal function with the same neurons...
But not sure how to interpretate these results either
[correlograms_internalfunction.pdf](https://github.com/SpikeInterface/spikeinterface/files/14874186/correlograms_internalfunction.pdf)

What type of metrics/extensions would you recommend to estimate connectivity? Is there a tutorial adressing the correlograms and/or connectivity?
Thanks!

zm711, 2024-04-04:
Could you do:

`sw.plot_autocorrelograms()` for each of those units so we can see what those look like? We still need to work more on our widgets documentation, so this is a good point that it is unclear.

alejoe91, 2024-04-04:
This looks like units with very few spikes. Can you check? :)

sofiapuvogelvittini, 2024-04-05:
Yes, that is true. There were very few spikes observed...
I experimented with another dataset containing neurons with around 500 spikes. I have attached a picture of the auto-correlograms, plotted using the spikeinterface internal function and also directly from the output of the sorting analyzer (the ccgs.npy file). 
[crosscorrelograms_internalfunction_individual_neurons.pdf](https://github.com/SpikeInterface/spikeinterface/files/14885764/crosscorrelograms_internalfunction_individual_neurons.pdf)

I am unsure how to interpret these plots. Should I expect a peak in the center, or do these plots appear normal?

Additionally, how would you recommend estimating connectivity between pairs of neurons based on the cross-correlograms? There are many papers out there, but I feel a bit unsure which guidelines to follow. Alternatively, would you suggest starting from the spike trains and checking their correlation to estimate connectivity? Do you have any tutorials addressing the types of analysis that could be performed based on the spikeinterface output?

Sorry for the numerous questions. I know there are many things that could be done with all the output from spikeinterface, but now that I am beginning to sort my data, I feel a bit lost regarding what to do with it.

Thank you very much  for your help and advice.

taningh86, 2024-04-05:
@sofiapuvogelvittini I am assuming you are having trouble with interpreting auto-correlograms/cross-correlograms? I have attached pictures of how my auto-correlograms and cross correlograms looks like. I will explain how it is generally interpreted in the field, but again there's not quite a standard because it depends on the type of neuron, its firing pattern and whatever intervention it is exposed to. Hope it makes sense.
[image]
[image]
[image]

Pictures with single plots are auto_correlograms. For unit 0 and unit 203 in the pictures you will see that at time 0 there's almost no peaks and the peaks increases to a certain time window (aorund 3-5 ms) and then goes down. What this is saying is that these neurons have a firing characteristic where the highest probability of firing or most spikes were seen with interspike interval of 3-5 ms and they also fire in longer timescales (>5 ms) but the probability decreases. This is good because, most neurons have a firing characteristic with some order or limits in their interspike interval (ISI). The fact that there's basically 0 spikes at time 0 is telling that this is a neuron (not a recording artifact) with a firing characteristic and not firing at random. A neuron with random firing will have spikes at all time scales. Auto-correlogram is basically superimposition of spike firing pattern of neuron 1 to itself and for a neuron with some order in their firing, this is what you expect. the autocorrelograms you showed in the other dataset is also great and it looks like it has specific firing patters as well. For your other data with flat plot, either something is wrong with your data or these neurons have very little to no firing.

If you're looking to establish connectivity between neurons that cross-correlogram is the way to go. However, it is still rudimentary and there are more advanced analysis there that is more robust in determining connectivity between neurons. the image with 4 plots is cross-correlogram between unit 0 and unit 206. You will see that when unit 0 is compared to unit 0 and same or unit 206 (green plots), the plot is exactly like its auto-correlogram. That totally makes sense. However, when unit 0 is compared to unit 206 and vice-versa (black plots) the peaks are almost opposite. The highest peak is at time 0 and it flattens out. The difference here is that, in cross-correlogram it is telling how far apart the spike times of unit 0 is from unit 206 and at what timepoint there's highest probability of spikes. With the peak at 0, it means that these neurons pretty much firing simultaneously and there's very little to 0 lag between their spiketimes for most spikes. this suggest that these neurons have coordinated firing and may be connected to each other via a direct or indirect synapse. 
Hope this makes sense. the interpretation of auto vs cross-correlogram is slightly different and easy to get confused.

sofiapuvogelvittini, 2024-04-05:
@taningh86, thanks a lot for your time. This is very helpfull!

sofiapuvogelvittini, 2024-04-09:
Hello, I have another question :). I have been reading that for estimating connectivity,  it is important to correct/normalize cross correlograms by the spiking frequency of the involved neurons. Are the cross-correlograms obtained with spikeinterface already normalized?

zm711, 2024-06-19:
Thanks for this wonderful discussion. I'll close this for now, but we can reopen if more issues come up.

closed 2024-06-19 by zm711