SpikeInterface/spikeinterface #2671: AttributeError: type object 'Kilosort4Sorter' has no attribute 'set_kilosort4_path'

mtehrani-code, 2024-04-04:
Hello again,

I'd like to use Kilosort4 within SI. I have it installed on my machine. Since it is MATLAB-based, I first set the path as follows:

kilosort_4_path = '/Users/matta/anaconda3/envs/kilosort/Lib/site-packages/'

Then:

si.Kilosort4Sorter.set_kilosort4_path(kilosort_4_path)

But I get this error:

AttributeError: type object 'Kilosort4Sorter' has no attribute 'set_kilosort4_path'

Could you please help me troubleshoot?

Thanks!
M

zm711, 2024-04-04:
@mtehrani-code,

Kilosort4 is python not matlab-based. So you just need to run:

```
import spikeinterface.sorters as ss
ss.run_sorter(sorter_name="kilosort4", recording=recording, output_folder=r'where/you/want/to/save'
```

Note that you need kilosort4 in the same environment as spikeinterface for this to work. If they are in different environments it won't work. So in this case you probably need to go into your kilosort environment and install spikeinterface (like we did in your other issue).

mtehrani-code, 2024-04-05:
@zm711 In Anaconda prompt, I'm in kilosort env.:

OK. So I will need to re-install SI? Will this automatically uninstall the currently installed SI?

I'm in the kilosort env. in Anaconda prompt:

(kilosort) C:\Users\matta>

Not sure what comes next in order to install SI, is it the cd documents/github/spikeinterface? and then conda activate spikeinterface_env?

Or will I not need to create another SI env. once I'm in the kilosort environment? And can go directly to pip install -e .[full,widgets]?

Sorry, a bit confused.

alejoe91, 2024-04-05:
You can just run `pip install spikeinterface[full,widgets]` to installed the latest released version

mtehrani-code, 2024-04-05:
Okay. Thanks.

I ran pip install spikeinterface[full,widgets] in the Kilosort env. 

However, when I run the si.installed_sorters() in Jupyter Notebook, I don't see kilosort on the list of installed sorters, even though I have it installed on my computer. This is what I get:

['simple', 'spykingcircus2', 'tridesclous2']

Then I get the message that it's not installed when I run (same thing when I try kilosort4)-- any ideas what could have gone wrong here?

import spikeinterface.sorters as ss
ss.run_sorter(sorter_name="kilosort", recording=recording_saved, output_folder=r'/Users/matta/Documents/openephys_recording')

Exception                                 Traceback (most recent call last)
Cell In[90], line 2
      1 import spikeinterface.sorters as ss
----> 2 ss.run_sorter(sorter_name="kilosort", recording=recording_saved, output_folder=r'/Users/matta/Documents/openephys_recording')

File ~\spikeinterface\src\spikeinterface\sorters\runsorter.py:175, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    168             container_image = singularity_image
    169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
--> 175 return run_sorter_local(**common_kwargs)

File ~\spikeinterface\src\spikeinterface\sorters\runsorter.py:222, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    219 SorterClass = sorter_dict[sorter_name]
    221 # only classmethod call not instance (stateless at instance level but state is in folder)
--> 222 output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
    223 SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
    224 SorterClass.setup_recording(recording, output_folder, verbose=verbose)

File ~\spikeinterface\src\spikeinterface\sorters\basesorter.py:105, in BaseSorter.initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
    101 @classmethod
    102 def initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder):
    103     # installed ?
    104     if not cls.is_installed():
--> 105         raise Exception(
    106             f"The sorter {cls.sorter_name} is not installed." f"Please install it with:  \n{cls.installation_mesg} "
    107         )
    109     if not isinstance(recording, BaseRecordingSnippets):
    110         raise ValueError("recording must be a Recording or a Snippets!!")

Exception: The sorter kilosort is not installed.Please install it with:  

To use Kilosort run:

        >>> git clone https://github.com/cortex-lab/KiloSort
    and provide the installation path by setting the KILOSORT_PATH
    environment variables or using KilosortSorter.set_kilosort_path().

    More information on KiloSort at:
        https://github.com/cortex-lab/KiloSort

alejoe91, 2024-04-05:
You need to specify `kilosort4`!

alejoe91, 2024-04-05:
What do toh get if you do `import kilosort`?

mtehrani-code, 2024-04-05:
I used the instructions here to install Kilosort4:
    https://github.com/MouseLand/Kilosort

mtehrani-code, 2024-04-05:
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[93], line 1
----> 1 import kilosort

ModuleNotFoundError: No module named 'kilosort'

> What do toh get if you do `import kilosort`?

ModuleNotFoundError                       Traceback (most recent call last)
Cell In[93], line 1
----> 1 import kilosort

ModuleNotFoundError: No module named 'kilosort'

zm711, 2024-04-06:
@mtehrani-code.

The kilosort4 instructions can be a bit confusing so definitely open an issue over on their repo to make sure you get it installed correctly. (Ideally you have a NVIDIA GPU for KS4), but like Alessio said the first step really needs to be getting KS4 installed if you want to use it.

alejoe91, 2024-04-11:
@mtehrani-code any update?

mtehrani-code, 2024-04-11:
@alejoe91 Hi! In the process of trying to get kilosort to work, I must have broken something because the tutorial no longer works. I get this error when I run import spikeinterface.full as si in Jupyter notebook:

ModuleNotFoundError                       Traceback (most recent call last)
Cell In[1], line 1
----> 1 import spikeinterface.full as si
      2 print(f"SpikeInterface version: {si.__version__}")

ModuleNotFoundError: No module named 'spikeinterface'

I'm not exactly sure what I did but I think I was trying to install kilosort in the same environment as spikeinterface. Then I decided to uninstall anaconda, then reinstall and try again. I also uninstalled and re-installed spikeinterface from source as Zach had explained before, but I keep getting the same error message.

I got a second opinion from someone who is experienced in coding/programming. He thinks the issue is that Jupyter isn't finding the spikeinterface_env. and we can't figure out why...

Currently, I have **spikeinterface_env** located here: C:\Users\matta\anaconda3\envs and **spikeinterface** located here C:\Users\matta\Documents\GitHub.

Also, I think my kilosort 4 installation is OK. I have all the system requirements for it and the installation seems to be done with no issues...All I can think of is that there is something wrong with the way files are located (environments? Paths?)

Pretty stuck at the moment :(

Could use all the help I can get!

zm711, 2024-04-11:
Hey @mtehrani-code,

You need to register an ipython kernel if you're using juypter in a different environment. See the [docs here](https://ipython.readthedocs.io/en/stable/install/index.html). It can be a little confusing to do that as a beginner.  I think what you need is to:

1) Have jupyter access
2) kilosort
3) spikeinterface

I would probably recommend a fresh install. So you would start like this:

```sh
conda create -n si_env_new python=3.10 jupyter
```
Their docs say 3.9 so if you want to be really safe you could do 3.9. and we might as well grab juypter while we are at it. Then you follow the rest of their instructions to get pytorch installed (ie conda activate si_env_new, pip uninstall.... etc).

Then once you're done getting kilosort installed I would recommend checking that you installed kilosort correctly with
```sh
conda activate si_env_new
python
>>> import kilosort
>>> import pytorch
>>> exit()
```
If an error occurs then you don't have kilosort or pytorch installed correctly and you need to work on that first. Once you have kilosort and jupyter all done
then you can do
```sh
cd path/to/spikeinterface/folder
pip install -e .[full,widgets]
```

then you can do the same
```sh
python
>>> import spikeinterface
>>> exit()
```
As long as that doesn't have an error then your environment is good to go and you can activate jupyter and use that (from inside the environment will work fine.

The reason why I named the `env` `si_env_new` is because you had previously installed `si_env` in a different issue. So if you haven't removed that environment you need to give this `env` a new name. If you've removed the old install of `si_env` then you can name this env `si_env` or `spikeinterface_env`.

mtehrani-code, 2024-04-15:
@zm711 Thank you!

I followed all these steps and I can import both spikeinterface and kilosort without error now. I have kilosort in my spikeinterface env.

I'm trying to register an ipykernel now using instructions here: https://ipython.readthedocs.io/en/stable/install/kernel_install.html

I have the kernel installed in my si environment and activated in jupyter notebook. However, I'm still getting the same error:
 
import spikeinterface.full as si
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[1], line 1
----> 1 import spikeinterface.full as si

Not sure what to try next...

mtehrani-code, 2024-04-17:
Turns out I needed to add **python -m** pip install -e .[full,widgets] to the code for conda to actually install the spikeinterface module in my environment! 

Looks like I'm able to run the tutorial in my separate environment now.

closed 2024-04-17 by mtehrani-code