SpikeInterface/spikeinterface #2672: multiple independent probes set with set_probegroup are not recognized by sorters

nexflatline, 2024-04-05:
I created 2 probes with 8 channels each and assigned them to a recording using `set_probegroup`.

When sorting the recording I got the error "There is no Probe attached to this recording. Creating a dummy one with contact positions".

All contacts in both probes have a channel assigned to them and the plotting and channel lists are correct. If I assign only one of the probes to the recording (`set_probe`), the sorters can recognize them.

Not sure if a bug or if I'm skipping some step (beginner here, sorry!).

chrishalcrow, 2024-04-05:
Hi @nexflatline , it would be help if you could supply some of your code, so we could easier find the problem. It could be something to do with the probegroup itself, or when it's attached to the recording.

One possibility: are you using the `in_place` argument in the set_probegroup function? You can either do the operation in place, something like:

`my_recording.set_probegroup(group_probe, in_place=True) `

Or you can create a new recording:

`recording_with_probe = recording.set_probegroup(group_probe)`

A few more details: https://spikeinterface.readthedocs.io/en/latest/modules/core.html?highlight=in_place#handling-probes.

Let me know if that helps, and if it doesn't please supply more code :)

nexflatline, 2024-04-08:
Hi @chrishalcrow ,

You are correct, I wasn't doing the operation in place and that solved the issue. My mistake!
Thank you very much :)

closed 2024-04-08 by nexflatline