SpikeInterface/spikeinterface #2688: Unsupported operand type(s) after upgrading from Kilosort 4.0 to 4.0.3

gkBCCN, 2024-04-10:
I'm using SpikeInterface to sort with Kilosort 4.0 and had success running the preprocessing, sorting, and analysis steps, but when I tried to open the results in Phy I got the dreaded QtWidgets error.  This error was fixed by the Kilosort team in version 4.0.3.  After reinstalling, if I run Kilosort from the GUI, everything works fine.  However, if I run it from the SpikeInterface pipeline, I get this error:

> File "/home/user/miniconda3/envs/spikeinterface/lib/python3.9/site-packages/kilosort/spikedetect.py", line 98, in template_centers
>     nx = np.round((xmax - xmin) / (dminx/2)) + 1
> TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'

I've encountered this error on multiple systems (Windows, Ubuntu) and with multiple datasets.

alejoe91, 2024-04-10:
Are there release notes somewhere for KS4?

gkBCCN, 2024-04-10:
Not that I know of.  I opened an issue in the KS GitHub.

gkBCCN, 2024-04-10:
Jacob added the release notes a couple hours ago: https://github.com/MouseLand/Kilosort/releases/tag/v4.0.3

zm711, 2024-04-10:
I know what the problem is (I think)!

They changed the default for dminx to be 32. Our default is None. @gkBCCN could you try running specifying for the dminx as 32 in spikeinterface.

PR #2701 (2024-04-10): Update ks4 wrapper for v4.0.3
Fixes #2688.

[KS4 Release Notes](https://github.com/MouseLand/Kilosort/releases/tag/v4.0.3)

We need to supply the `dminx` (KS default is now 32). Supplying None will break new versions. But supplying 32 to old versions shouldn't hurt anything so this should be a clean fix even if people use older versions (4.0.2 or 4.0.1).
closed 2024-04-10 by alejoe91