SpikeInterface/spikeinterface #2696: `n_jobs` not being respected between set_global_kwargs and running sorter

zm711, 2024-04-10:
As discussed in #2690,

[image]

alejoe91, 2024-04-10:
The problem is this: 

In the `basesorter.py` we have a [`default_job_kwargs`](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/basesorter.py#L24) that by default sets `n_jobs` to -1 ([here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/basesorter.py#L24)). My suggestion is to remove this and `warn` if the sorter `requires_binary_data` and `n_jobs=1`, saying that this might make things slower.

Another option (and maybe complementary) is to set default n_jobs globally to something like 0.5.

What do you think @zm711 ?

zm711, 2024-04-10:
> In the basesorter.py we have a [default_job_kwargs](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/basesorter.py#L24) that by default sets n_jobs to -1 [...]

I think this sounds right! I don't think we should change the n_jobs without the user knowing!

> Another option (and maybe complementary) is to set default n_jobs globally to something like 0.5.

Yeah I'm not sure what the fair value for this would be. Like 0.9 or 0.8 would probably be safe to leave a core or two behind whereas 0.5 would leave more stuff behind. But yeah I think changing the default from -1 would be nice.