SpikeInterface/spikeinterface #2697: kilosort4: skip_kilosort_preprocessing

sachuriga283, 2024-04-10:
Dear,

I have trying to run some test on kilosort4, and since I'm not using neuropixel so I would like to skip the preprocess of kilosort. 
But I'm wondering, does `skip_kilosort_preprocessing` skip the high-pass filtering, CAR, and whitening steps?

best,

zm711, 2024-04-10:
Hey @sachuriga283,

Here are all the params:
https://github.com/SpikeInterface/spikeinterface/blob/bbe871993954fde757034bbab98f7c4c788ead7d/src/spikeinterface/sorters/external/kilosort4.py#L57-L91

to skip CAR you would need to do: `do_CAR=False`

To skip highpass filtering and whitening you would do `skip_kilosort_preprocesing`.

sachuriga283, 2024-04-10:
OK, thanks

zm711, 2024-04-10:
Feel free to open new issues if you run into any problems :)

closed 2024-04-10 by zm711