SpikeInterface/spikeinterface #2699: kilosort4 runtime error with skiping the drift correction

sachuriga283, 2024-04-10:
Dear,

I'm testing on the kilosrt4 with 6shank 64 channel probe data. When I skipping the drift correction the kilosor4 stopped with error:

` Skipping kilosort preprocessing.
Skipping drift correction.

'Extracting spikes using templates
Re-computing universal templates from data.
Traceback (most recent call last):
  File "/cluster/home/sachur/Overlap_project/code/Python/Kilosort/kilosort4_multfile.py", line 188, in <module>
    sorting = ss.run_sorter_by_property('kilosort4',
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/launcher.py", line 297, in run_sorter_by_property
    sorting_list = run_sorter_jobs(job_list, engine=engine, engine_kwargs=engine_kwargs, return_output=True)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/launcher.py", line 106, in run_sorter_jobs
    sorting = run_sorter(**kwargs)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 175, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 225, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 293, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/external/kilosort4.py", line 260, in _run_from_folder
    st, tF, _, _ = detect_spikes(ops, device, bfile, tic0=tic0, progress_bar=progress_bar)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/run_kilosort.py", line 392, in detect_spikes
    st0, tF, ops = spikedetect.run(ops, bfile, device=device, progress_bar=progress_bar)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/spikedetect.py", line 193, in run
    ops['wPCA'], ops['wTEMP'] = extract_wPCA_wTEMP(
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/spikedetect.py", line 54, in extract_wPCA_wTEMP
    X = bfile.padded_batch_to_torch(j, ops)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/io.py", line 666, in padded_batch_to_torch
    X = super().padded_batch_to_torch(ibatch)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/io.py", line 494, in padded_batch_to_torch
    X = torch.zeros((self.n_chan_bin, self.NT + 2*self.nt), device=self.device)
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.'`

And the paras for sorter is like below,
`  para_ks4 = {'skip_kilosort_preprocessing': True,
                    'nblocks': 0,
                    'nearest_templates': 9,
                    'min_template_size': 16.5,
                    'batch_size': 320000,
                    'do_correction':False}

  sorting = ss.run_sorter_by_property('kilosort4',
                                      recording=recording_f0,
                                      working_folder=base_folder / ks3,
                                      grouping_property='group',
                                      mode_if_folder_exists='keep', **para_ks4)`

Would you think this could be a error from spikeinterface or kilosort4?

best,

zm711, 2024-04-10:
What happens if you type 

```python
import torch

print(torch.cuda.is_available())
```

A cuda error like this is likely KS4 because their code is where the cuda calls are made.

sachuriga283, 2024-04-10:
Hey I have tested the code, its giving me `True` .

and if I set the `nblock=1` and  `do_correction=True` the sorter is works.

```bash
Loading recording with SpikeInterface...
number of samples: 38698575
number of channels: 10
numbef of segments: 1
sampling rate: 30000.0
dtype: int16
========================================
Skipping kilosort preprocessing.

computing drift
Re-computing universal templates from data.

  0%|          | 0/121 [00:00<?, ?it/s]
  1%|          | 1/121 [00:03<06:16,  3.14s/it]
  2%|▏         | 2/121 [00:04<03:59,  2.02s/it]
  2%|▏         | 3/121 [00:05<03:15,  1.66s/it]
  3%|▎         | 4/121 [00:06<02:54,  1.49s/it]
  4%|▍         | 5/121 [00:08<02:41,  1.40s/it]
  5%|▍         | 6/121 [00:09<02:44,  1.43s/it]
  6%|▌         | 7/121 [00:10<02:35,  1.36s/it]
  7%|▋         | 8/121 [00:12<02:29,  1.32s/it]
  7%|▋         | 9/121 [00:13<02:24,  1.29s/it]
  8%|▊         | 10/121 [00:14<02:20,  1.27s/it]
  9%|▉         | 11/121 [00:15<02:18,  1.26s/it]
 10%|▉         | 12/121 [00:17<02:34,  1.41s/it]
 11%|█         | 13/121 [00:18<02:26,  1.36s/it]
 12%|█▏        | 14/121 [00:19<02:20,  1.32s/it]
 12%|█▏        | 15/121 [00:21<02:16,  1.29s/it]
 13%|█▎        | 16/121 [00:22<02:13,  1.27s/it]
 14%|█▍        | 17/121 [00:23<02:10,  1.26s/it]
 15%|█▍        | 18/121 [00:24<02:08,  1.25s/it]
 16%|█▌        | 19/121 [00:26<02:06,  1.24s/it]
 17%|█▋        | 20/121 [00:27<02:04,  1.23s/it]
 17%|█▋        | 21/121 [00:28<02:03,  1.23s/it]
 18%|█▊        | 22/121 [00:29<02:01,  1.23s/it]
 19%|█▉        | 23/121 [00:32<02:32,  1.56s/it]
 20%|█▉        | 24/121 [00:33<02:21,  1.46s/it]
 21%|██        | 25/121 [00:34<02:13,  1.39s/it]
 21%|██▏       | 26/121 [00:35<02:06,  1.33s/it]
 22%|██▏       | 27/121 [00:36<02:02,  1.30s/it]
 23%|██▎       | 28/121 [00:38<01:58,  1.28s/it]
 24%|██▍       | 29/121 [00:39<01:56,  1.26s/it]
 25%|██▍       | 30/121 [00:40<01:53,  1.25s/it]
 26%|██▌       | 31/121 [00:41<01:51,  1.24s/it]
 26%|██▋       | 32/121 [00:43<01:50,  1.24s/it]
 27%|██▋       | 33/121 [00:44<01:48,  1.23s/it]
 28%|██▊       | 34/121 [00:45<01:47,  1.23s/it]
 29%|██▉       | 35/121 [00:46<01:45,  1.23s/it]
 30%|██▉       | 36/121 [00:47<01:44,  1.23s/it]
 31%|███       | 37/121 [00:49<01:42,  1.23s/it]
 31%|███▏      | 38/121 [00:50<01:41,  1.23s/it]
 32%|███▏      | 39/121 [00:51<01:40,  1.23s/it]
 33%|███▎      | 40/121 [00:52<01:39,  1.22s/it]
 34%|███▍      | 41/121 [00:54<01:37,  1.22s/it]
 35%|███▍      | 42/121 [00:55<01:36,  1.22s/it]
 36%|███▌      | 43/121 [00:56<01:35,  1.22s/it]
 36%|███▋      | 44/121 [00:57<01:34,  1.22s/it]
 37%|███▋      | 45/121 [01:01<02:23,  1.88s/it]
 38%|███▊      | 46/121 [01:02<02:06,  1.69s/it]
 39%|███▉      | 47/121 [01:03<01:54,  1.55s/it]
 40%|███▉      | 48/121 [01:04<01:45,  1.45s/it]
 40%|████      | 49/121 [01:06<01:39,  1.38s/it]
 41%|████▏     | 50/121 [01:07<01:34,  1.33s/it]
 42%|████▏     | 51/121 [01:08<01:30,  1.30s/it]
 43%|████▎     | 52/121 [01:09<01:27,  1.27s/it]
 44%|████▍     | 53/121 [01:10<01:25,  1.26s/it]
 45%|████▍     | 54/121 [01:12<01:23,  1.25s/it]
 45%|████▌     | 55/121 [01:13<01:21,  1.24s/it]
 46%|████▋     | 56/121 [01:14<01:20,  1.23s/it]
 47%|████▋     | 57/121 [01:15<01:18,  1.23s/it]
 48%|████▊     | 58/121 [01:17<01:17,  1.23s/it]
 49%|████▉     | 59/121 [01:18<01:16,  1.23s/it]
 50%|████▉     | 60/121 [01:19<01:14,  1.22s/it]
 50%|█████     | 61/121 [01:20<01:13,  1.22s/it]
 51%|█████     | 62/121 [01:21<01:12,  1.22s/it]
 52%|█████▏    | 63/121 [01:23<01:10,  1.22s/it]
 53%|█████▎    | 64/121 [01:24<01:09,  1.22s/it]
 54%|█████▎    | 65/121 [01:25<01:08,  1.22s/it]
 55%|█████▍    | 66/121 [01:26<01:07,  1.22s/it]
 55%|█████▌    | 67/121 [01:28<01:06,  1.22s/it]
 56%|█████▌    | 68/121 [01:29<01:04,  1.22s/it]
 57%|█████▋    | 69/121 [01:30<01:03,  1.22s/it]
 58%|█████▊    | 70/121 [01:31<01:02,  1.22s/it]
 59%|█████▊    | 71/121 [01:32<01:01,  1.22s/it]
 60%|█████▉    | 72/121 [01:34<00:59,  1.22s/it]
 60%|██████    | 73/121 [01:35<00:58,  1.22s/it]
```

sachuriga283, 2024-04-10:
Hey, 

the log file above is generated with my code like this, and I have added the code you proivided thats showing `True` in the log file.

```bash
para_ks4 = {'skip_kilosort_preprocessing': True,
                    'nblocks': 1,
                    'nearest_templates': 9,
                    'min_template_size': 16.5,
                    'batch_size': 320000,
                    'do_correction':True}

sorting = ss.run_sorter_by_property('kilosort4',
                                    recording=recording_f0,
                                    working_folder=base_folder / ks3,
                                    grouping_property='group',
                                    mode_if_folder_exists='keep', **para_ks4)
```

the full log will be looks like this, there happens a another error...but at least it step to the clustering process..

```bash
Wed Apr 10 20:04:54 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.23.06              Driver Version: 545.23.06    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  Tesla P100-PCIE-16GB           On  | 00000000:03:00.0 Off |                    0 |
| N/A   39C    P0              36W / 250W |      0MiB / 16384MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+
/cluster/apps/eb/software/Python/3.10.4-GCCcore-11.3.0/lib/python3.10/site-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
True
Current working directory before
/cluster/home/sachur/Overlap_project/data/OpenEphys

65588_2024-03-08
65588_2024-03-08_16-10-29_C
Current working directory before
/cluster/home/sachur/Overlap_project/data/OpenEphys

Channel ids: ['CH1' 'CH2' 'CH3' 'CH4' 'CH5' 'CH6' 'CH7' 'CH8' 'CH9' 'CH10' 'CH11'
 'CH12' 'CH13' 'CH14' 'CH15' 'CH16' 'CH17' 'CH18' 'CH19' 'CH20' 'CH21'
 'CH22' 'CH23' 'CH24' 'CH25' 'CH26' 'CH27' 'CH28' 'CH29' 'CH30' 'CH31'
 'CH32' 'CH33' 'CH34' 'CH35' 'CH36' 'CH37' 'CH38' 'CH39' 'CH40' 'CH41'
 'CH42' 'CH43' 'CH44' 'CH45' 'CH46' 'CH47' 'CH48' 'CH49' 'CH50' 'CH51'
 'CH52' 'CH53' 'CH54' 'CH55' 'CH56' 'CH57' 'CH58' 'CH59' 'CH60' 'CH61'
 'CH62' 'CH63' 'CH64']
Sampling frequency: 30000.0
Number of channels: 64
Number of segments: 1
Traces shape: (60000, 64)
ASSY-236-F - cambridgeneurotech - 64ch - 6shanks
['is_filtered', 'probe_0_planar_contour']
write_binary_recording with n_jobs = 40 and chunk_size = 150000

write_binary_recording:   0%|          | 0/258 [00:00<?, ?it/s]
write_binary_recording:   0%|          | 1/258 [00:05<24:57,  5.83s/it]
[... 372 lines cut ...]
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/launcher.py", line 106, in run_sorter_jobs
    sorting = run_sorter(**kwargs)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 175, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 225, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 293, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/spikeinterface/sorters/external/kilosort4.py", line 260, in _run_from_folder
    st, tF, _, _ = detect_spikes(ops, device, bfile, tic0=tic0, progress_bar=progress_bar)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/run_kilosort.py", line 409, in detect_spikes
    st, tF, ops = template_matching.extract(ops, bfile, Wall3, device=device,
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/template_matching.py", line 36, in extract
[... cut ...] =device)
  File "/cluster/home/sachur/.local/lib/python3.10/site-packages/kilosort/template_matching.py", line 155, in run_matching
    st[k:k+nsp, 0] = iX[:,0]
RuntimeError: The expanded size of the tensor (1368) must match the existing size (2255) at non-singleton dimension 0.  Target sizes: [1368].  Tensor sizes: [2255]

Spike sorting failed. You can inspect the runtime trace in /cluster/home/sachur/Overlap_project/data/OpenEphys/65588_2024-03-08_16-10-29_C_ks4/0/spikeinterface_log.json.

```

alejoe91, 2024-04-10:
Yeah Inhad troubles running KS4 without whitening! So whitening is definitely required!

sachuriga283, 2024-04-10:
OK! that's good to know, let me try with `whiten`

sachuriga283, 2024-04-11:
OK, I'm gonna close this issue. The above error was happens in the cluster PC, once I run the code in Local PC the code is working perfectly no matter I'm `'do_CAR':False` or `'skip_kilosort_preprocessing':True`. I the local PC I created the env that recommended by kilosort4 which is `python==3.9` and `cuda==11.8`. However, with the cluster PC, I don't have as many options. The cluster environment uses`.  So, my guess is that some conflicts occur on the cluster PC when running Kilosort4. Later, I will try to run Kilosort4 with Singularity. That should solve the problem.

```bash
34 clusters found, in  4.57s; total  28.44s

Merging clusters
32 units found, in  0.10s; total  28.55s

Saving to phy and computing refractory periods
11 units found with good refractory periods

Total runtime: 28.65s = 00:00:28 h:m:s
========================================
Loading recording with SpikeInterface...
number of samples: 38698575
number of channels: 10
numbef of segments: 1
sampling rate: 30000.0
dtype: int16
========================================
Preprocessing filters computed in  0.11s; total  0.11s
Skipping drift correction.

Extracting spikes using templates
Re-computing universal templates from data.
```

closed 2024-04-11 by sachuriga283