SpikeInterface/spikeinterface #2702: Coverting .rhs to edf format

kshtjkumar, 2024-04-10:
Hi, i am trying to run some available seizure scoring and detection algorithms using Neuroscore and other methods. Most of these use .edf as input file. So I tried converting my intan .rhs files to .edf. I followed this method:

loaded trace of each channel to a list using spikeinterface 
converted the list to a dataframe
using MNE converted this dataframe to MNE raw object 
then tried exporting this raw object to .edf format

I was wondering if there is a better method to do so using spike interface ?

zm711, 2024-04-10:
As far as I know we don't have an export to `.edf` option any where in the code. I'll tag @samuelgarcia and @alejoe91 so they can confirm.

alejoe91, 2024-04-11:
You're right @zm711 
We have no converter on our side! Your approach looks ok @kshtjkumar, as long as all traces can fit in memory. Unfortunately, there is currently no better alternative