SpikeInterface/spikeinterface #2703: Jupyter Notebook imports hanging

TheAnton205, 2024-04-10:
Hello, I had spikeinterface fully up and working before and decided to upgrade it to version 0.101.0 for new functionality. However, things went downhill and am looking for help. Here is the steps of what happened:
1. Upgraded to 0.101.0 from 0.99.1
    i. Running command "read_openephys" gave error ```SYNC channel is not present in the recording. Set load_sync_channel to False``` 
    despite not needing it before. I did what it said, inputtng the argument load_sync_channel to False and got the exact same error 
    message.
2. Realized it was the version messing it up, downgraded back to 0.99.1
    i. Same issue with SYNC Channel occurred
3. Pip uninstalled every spikeinterface and reinstalled through anaconda terminal (which runs the notebook), it installed version 0.100.5
    i. New issue occurs, cannot even import what I need. When running ```import spikeinteface.widgets as sw```, the import just hangs. 
       Additionally, I had ```import spikeinterface.full as si```, which also hung but I changed to ```spikeinterface.core```. Below are all of my 
       imports to which everything but the two I just mentioned working fine
4. Decided to downgrade to 0.99.0, which is what it is currently running on in case I wrote down the wrong version number.
    i. I get the same issue as step 3.
```
import spikeinterface.core as si #Originally spikeinterface.full, changed to get rid of it getting stuck
import pywidgets
import ipympl
import spikeinterface
import spikeinterface.sorters as ss
from spikeinterface.sorters import run_sorter
import spikeinterface.widgets as sw #<------------------------------ import begins to hang and cell never reaches end.
import spikeinterface.extractors as se
from spikeinterface.preprocessing import common_reference
#from spikeinterface import create_sorting_analyzer
import binascii

import matplotlib.pyplot as plt
import seaborn as sns
import numpy as np #version == 1.20
```

Thanks!

zm711, 2024-04-10:
So pip uninstalling spikeinterface won't get also remove the underlying dependencies (ie the version of neo you have would not downgrade at the same time). I bet the `read_openephys` change occurred because upgrading to Neo. It might be worth just starting a fresh environment if you want to downgrade. I forget if @alejoe91 or @h-mayorquin did the recent update of openephys here so I'll let them comment here before we dive too deep into a rabbit hole.

samuelgarcia, 2024-04-11:
For this kind of testing you really should have diffrents envs!

zm711, 2024-07-11:
@TheAnton205 totally missed this. This could be an issue with open_ephys and your data. You could try `load_sync_channel=False`, and if that doesn't work open an issue on Neo repo, [here](https://github.com/NeuralEnsemble/python-neo)