SpikeInterface/spikeinterface #2711: Update `SortingAnalyzer` Docs

zm711, 2024-04-12:
We need to:

* [ ] Remove references to `fast_templates` in core.rst and in tutorials
* [ ] Add section to explain operators. ie get waveforms for more complex operators or just get avg without waveforms
* [ ] Add mention of delete child extensions if recompute parrent (in the core docs)--already done in tutorials
* [ ] Have a section that clearly explains the differences between `save` within `compute` and `save_as` as a `SortingAnalyzer` function
* [ ] Make sure parent-child is clear in core docs (already done in tutorials-may need to update if fast-template in tutorial)
* [ ] Explain the arguments/input types for `.compute` in core docs (& ?tutorial)

Let me know if I'm missing anything or just edit the list yourself :) 

@chrishalcrow do you want to work on this and have me edit or vice-versa?

chrishalcrow, 2024-04-12:
Hello,
I'm happy to work or to edit. If I work on it, I'll have some time early next week :)

Before doing anything, I think it's helpful to clarify and agree what should be on each documentation page so we don't repeat ourselves too much. Here's a list of all the pages in the docs about SortingAnalyzer, and roughly what's in them:

**Module documentation / core / Sorting**
How to make, save and load a SortingAnalyzer
Basic compute, for core computes.

**Module documentation / PostProcessing**
How to find out which extensions are loaded and loadable
The PostProcessing extensions

**Module examples / SortingAnalyzer**
Nice example of making a SortingAnalyzer and computing some things

**How_to / get_started**
Intro tutorial, does a few things

**How_to / From waveform extractor to sorting analyzer**
A dictionary for how to convert from a waveform extractor to a sorting analyzer

To me it's not obvious where to put the parent-child stuff and the "advanced" compute stuff. It feels core-ish, but it's only interesting when there are more examples around (in Postprocessing/Qualitymetrics). We could explain the concepts in core, then have examples in both the Postprocessing and Quality metrics sections and link back to core?

Putting examples and discussion of these topics in the **Module examples / Sorting Analyzer** seems like a good idea. And updating the how to's is obviously a good idea.

zm711, 2024-04-12:
> Before doing anything, I think it's helpful to clarify and agree what should be on each documentation page so we don't repeat ourselves too much.

I think repeating ourselves might be beneficial. Some users want to read the docs others want to do tutorials/how tos. So having the same info in both places makes sense  to me. I think how deep different sections should go though is an important point. If something has a particular goal (convert wvf_extractor -> sorting_analyzer, shouldn't dive deep into internal mechanisms of `SortingAnalyzer`). 

> To me it's not obvious where to put the parent-child stuff and the "advanced" compute stuff. It feels core-ish, but it's only interesting when there are more examples around (in [...]

this seems good to me. Some mention of parent-child should happen in `core` to make it clear that this is a fundamental part of the a `SortingAnalyzer` along with the format and saving options. We have an issue where an end-user was quite upset with how one of the saving defaults worked so I think we need to be clearer about when spikeinterface saves something and when it lives just in memory.

> Putting examples and discussion of these topics in the Module examples / Sorting Analyzer seems like a good idea. 

This seems like a key place to have some more explanations of the ideas behind the analyzer so agree!

> How_to / get_started
> Intro tutorial, does a few things

For this one I personally would just want to make sure we are not wrong about anything rather than too deep of an explanation, but if you feel like it's super valuable to have deep explanations here I could be convinced.

chrishalcrow, 2024-04-15:
@zm711 All sounds sensible. I like the "depth" way of thinking! Will make a start

chrishalcrow, 2024-04-16:
>  Add section to explain operators. ie get waveforms for more complex operators or just get avg without waveforms

@zm711 Hello, not sure that _I_ understand operators enough to explain them! Is this the idea that for `templates` if you use `average`  or `std` you only need `random_spikes`. But for `percentile` you need `waveforms`. Are there other examples? Thanks!

zm711, 2024-04-16:
That's specifically what I was referring to because we deleted `fast_templates` I wanted to make sure we documented what the difference between the old `fast_templates` and the current merged `templates`. I don't think there is currently any other thing that needs that super fine granularity. But templates are special.

chrishalcrow, 2024-04-17:
> That's specifically what I was referring to because we deleted `fast_templates` I wanted to make sure we documented what the difference between the old `fast_templates` and the current merged [...]

Great! I think you're better placed to add something about this (I hadn't really noticed that `fast_templates` existed until it was deleted). If you could add a paragraph, probably at the very end of the `SortingAnalyzer` section in core.rst, that'd be fab. Thanks!

alejoe91, 2024-04-29:
I think we can close this! Thanks @chrishalcrow and @zm711 <3

closed 2024-04-29 by alejoe91