SpikeInterface/spikeinterface #2723: Full spikerinterface tests codecov failing

chrishalcrow, 2024-04-16:
Our full spikeinterface tests codecovs are failing. I’ve found the problem, and a bad solution.

Here’s an example of a failing run: https://github.com/SpikeInterface/spikeinterface/actions/runs/8688996887

The steps that fails is “run tests”. This is suprising, because “Full spikeinterface tests” succeeds (see https://github.com/SpikeInterface/spikeinterface/actions/workflows/full-test.yml ). The difference between them is that “Full spikeinterface tests” test module by module. While “run tests” in “Full spikeinterface tests codecov” run a full pytest. So the error must be something leaking from one set of module tests into another. So this is really the issue.

I’ve narrowed down the moment the issue rears it’s head and figured out a hacky solution. This is helpful, but doesn’t get to the heart of the issue, so I don’t think we should merge this into main. I’ll explain where the error is next.

There are 7 test FAILURES such as 
```
src/spikeinterface/postprocessing/tests/test_unit_localization.py::UnitLocationsExtensionTest::test_extension - ValueError: Computing templates with operators ('percentile', 1) needs the 'waveforms' extension
```
You can re-create a similar error message by doing 
```
… # make a sorting_analyzer
sorting_analyzer.compute(“random_spikes”)
sorting_analyzer.compute(“templates”, operators=[[‘percentile’, 1]])
```
This isn’t a problem in itself, but the test is assuming that the templates compute is using default kwargs (these are `operators=["average", "std”]`, which only depend on “random_spikes” not “waveforms”).

So, the key questions is: why isn’t `sorting_analyzer.compute(“templates”)` using the default kwargs? This occurs at [line 98](https://github.com/SpikeInterface/spikeinterface/blob/fa57feeca6b403ae8595204f18313f038f7dc5ac/src/spikeinterface/postprocessing/tests/common_extension_tests.py#L98) of postprocessing/tests/common_extension_tests.py and [line 19](https://github.com/SpikeInterface/spikeinterface/blob/fa57feeca6b403ae8595204f18313f038f7dc5ac/src/spikeinterface/sortingcomponents/tests/test_template_matching.py#L19) of sortingcomponents/tests/test_template_matching.py . You can hack a solution by explicitly adding `operators = ["average”]` to the tests. See my main for an example, and see here (https://github.com/chrishalcrow/spikeinterface/actions/runs/8701398446/job/23863321543) to see that “run tests” is successful (the whole thing isn’t successful on my branch cause I’ve not got my tokens sorted).

Phew! Any ideas?

zm711, 2024-04-16:
Could this be some cache for the tests? The normal tests do a module by module test right? So each module gets its own pytest process. But for full we just run them all together. So I wonder if the sortinganalyzer object because it is just living in memory gets corrupted between the different module testing? I don't see why it would happen, but I honestly don't know some of the guts of how pytest works (and pytest 8+ has had some issues as Heberto and I discussed, but not sure if that could lead to this).

alejoe91, 2024-04-16:
Or maybe some fixture at the session scope? I'll take a look!

zm711, 2024-04-30:
Tests are working as of #2777 and I checked today's and they passed :)

closed 2024-04-30 by zm711